opendata-stuttgart/sensors-software #289: Auto-update progress indication

bertrik, 2018-10-28:
While programming a batch of nodemcu's with the airrohr-firmware an auto-update happens during the setup process. Could we add a kind of indication that the auto-update is happening? For example a couple more debug messages in the serial output log.

ricki-z, 2018-10-28:
In the serial output this should be shown. In the default debug level you should see the messages:
```
Starting OTA update ...
NodeMCU firmware :  (+ the actual version)
```
if there is an update the update will be downloaded and installed and the NodeMCU will restart. While this is running we can't write anything to the serial output log.

bertrik, 2018-10-28:
Ok, clear, I'll close this issue

closed 2018-10-28 by bertrik