opendata-stuttgart/sensors-software #297: Add Support for AMS CC811

raspyweather, 2018-11-02:
Please add support for this cheap air quality sensor

Adorfer, 2018-11-08:
What does this sensor actually measure? 
From the PDF i deduct, it's 
1) CO2 in the range 400ppm up to 32768ppm
2) "equivalent Total Volatile Organic Compound" (eTVOC) in the range 0ppb up to 32768ppb. 

https://ams.com/documents/20143/36005/CCS811_DS000459_6-00.pdf/c7091525-c7e5-37ac-eedb-b6c6828b0dcf

ropeters68, 2018-11-08:
That would be a very nice feature!

ricki-z, 2018-11-09:
This sensor uses some kind of 'auto calibration' using the lowest measured value in the last 24 hours as a baseline. So you can't compare two sensor places because they may have very different 'baselines'.

ricki-z, 2018-11-09:
This is a problem for the BME680 also.
One possible solution: do all CC811 (or BME680) in one room for 24 hours, save baseline, repeat this every 3 months. But who should do this?

ropeters68, 2018-11-09:
You can't compare the values even if they have the same baseline because it's an aspecific sensor (you measure something but you don't know what). However you can use it to detect pollution incidents. These type of sensors are widely used in E-Nose projects (like in the Port of Rotterdam and near chemical plants) to detect bad smells and spills. So although the measurments are not directly comparable they do provide valuable data.
----Origineel Bericht----

raspyweather, 2018-11-10:
TVOC can measure stuff like fumes from paint, chemicals, gasoline etc.. 
You should read this with a grain of salt, but that's a very interesting source about VOCs IMHO:
http://key-vocs.eu/sites/default/files/ENV56%20Final%20Publishable%20Summary_accepted.pdf 

Of course, I'm not a professional in Chemistry and I'm still getting used to measuring my environment and trying to get to know which values could have which background. 

I would love to see the ability to simply plug such a sensor in my current air quality station and see what's it recording without having to get another µC and another hacked together firmware.

ricki-z, 2018-11-17:
But how useful is your measurement if you can't say what exactly you have measured? With that sensor you can't decide if it's a broken wodka bottle (thrown away by young people on their way to a party) or from a chemical plant near your home?

raspyweather, 2018-11-17:
How useful is a measurement of particulate matter if you can't decide whether it's a someone smoking or from the exhaust of a car?
The point of measuring your environment is monitoring not necessary determining the 'guilty' person behind each peak, instead you try to understand your environment by measuring data and making statistical analysis.
And it can even be a valid result that you might have a bunch of smoking and drinking young people in your neighborhood, is it an interesting result? Not spectacular, but it can be a result. As you need to take every measurement with a grain of salt and care before making blind guesses about maxima and minima in data, you need to do it with environmental data