opendata-stuttgart/sensors-software #307: Problem with basic auth in config mode

ricki-z, 2018-11-07:


ricki-z, 2018-11-07:
Expected: browser should ask for username and password
Actual: white page after timeout, no request for auth

Informatic, 2018-11-07:
Both current `latest.bin` and `latest_beta.bin` seem to be working fine on both Chromium 70 and Firefox 63.
https://d.inf.re/b/e824ab194946b183b1e99d6433238af7.png

fmeerkoetter, 2018-11-07:
I've tested a built from the latest state of the beta branch. It worked from a Linux laptop (with Chrome). Will try an Android phone.

fmeerkoetter, 2018-11-07:
[image]
Also seems to work on my Android phone.

Hmmmm. The only thing I noticed that the Captive Portal detection didn't work (I assume the captive portal doesn't like basic auth)?

ricki-z, 2018-11-07:
But captive portal should work if possible. We have more and more non-techie users ;-)

fmeerkoetter, 2018-11-07:
I've also retested with Chrome on android (the former screenshot was with Firefox on Android).

fmeerkoetter, 2018-11-07:
> But captive portal should work if possible. We have more and more non-techie users ;-)

But _can_ it work if basic auth is on? I have no idea. Checking...

fmeerkoetter, 2018-11-07:
> > But captive portal should work if possible. We have more and more non-techie users ;-)
> 
> But _can_ it work if basic auth is on? I have no idea. Checking...

Just to be sure I tested if the captive portal is working correctly on Android with the basic auth == off. The result is it does. *puh*

ricki-z, 2018-11-07:
The captive portal is working with MacOS if basic auth is activated.
But Android seems to have a problem (tested with a smartphone and a tablet with Android 7.x). 
A check with an Apple device and with Windows (7 and 10) would be nice.

ricki-z, 2018-11-08:
I have checked this with the actual release version. This version shows the same behaviour, there is no request for username and password on Andriod and the config page remains empty.

fmeerkoetter, 2018-11-08:
@ricki-z Windows10 works fine. Captive portal mode is triggered and a browser with the auth dialog comes up.

LuchtwachtersDelft, 2018-11-09:
Besides, the basic auth popup can also be scary to non-techie users. How about building a friendly login screen into the firmware rather than relying on basic auth?

ricki-z, 2018-11-09:
This logon is not the default, so most users won't see this. This is used by some people that have connected the sensor to an open wifi network.
Building a 'new' login screen needs resources we don't have (memory).