opendata-stuttgart/sensors-software #323: Increase maximum sketch size to 1 MB

LuchtwachtersDelft, 2018-11-18:
I know the first thing you want to say is `No because OTA`. But hear me out.

When compiling the firmware you will something like this:

```
Compiling debug version of 'airrohr-firmware' for 'NodeMCU 1.0 (ESP-12E Module)'
Program size: 515,888 bytes (used 49% of a 1,044,464 byte maximum) (46.43 secs)
```

The maximum program size is 1 MB. Because of that and an equal amount of storage needed for Over The Air update we target 0.5 MB. 

But that is a conflation of 2 limitations.

The ESP8266 has a 4 MB flash. It can be divided in several ways: 

- 1 MB sketch, 3 MB SPIFFS
- 2 MB sketch, 2 MB SPIFFS
- 3 MB sketch, 1 MB SPIFFS

I have configured it as 3 MB sketch, 1 MB SPIFFS. Yet after compiling, `(used 49% of a 1,044,464 byte maximum) `. That is correct. Even though the storage for sketches is 3 MB, the limitation of 1 MB per program remains. 

However, that means we can use 1 MB sketches and reserve 1 MB for OTA, and still have 2 MB left for SPIFFS. 

How much SPIFFS space do we use? When compiled with ` 2 MB sketch, 2 MB SPIFFS`:

```C
DEBUG_MSG("Total bytes: %d\n", fs_info.totalBytes);
DEBUG_MSG("Used bytes: %d\n", fs_info.usedBytes);
DEBUG_MSG("Page size: %d\n", fs_info.pageSize);		

Total bytes: 1953282
Used bytes: 3765
Page size: 256
```

So we can use `2 MB sketch, 2 MB SPIFFS` or even `3 MB sketch, 1 MB SPIFFS`, increase the maximum sketch size to 1 MB instead of 0.5 MB and still have plenty of unused SPIFFS.

ricki-z, 2018-11-18:
You can do this for a newly installed NodeMCU. If you flash such an image over an existing firmware you will loose your configuration (change in location of SPIFFS headers) and need to reconfigure the sensor.
For a already flashed NodeMCU such a change is not possible via OTA. So we would need to newly flash all 7000 actually active sensors.

ricki-z, 2018-11-26:
But for this we need to flash two times. Steps for this process:
Download new image to SPIFFS
Flash small firmware that only flashes the version saved to SPIFFS.
Reboot to start the small firmware image, this then copies the new firmware to flash memory.
Reboot to new firmware version.

This adds some complexity, that we avoid with the direct flashing. Actually the firmware is downloaded and flashed, if this was finished okay. If the the firmware can't be downloaded everything will work as until then. With all those steps above we have too many steps that can go wrong.

DeeKey, 2019-02-07:
Fact is - sketch size of 0,5 MB is a limiting factor for now. In many issues I see a note: "will not do, as it will increase scetch footprint. Every bit matters" 
On the other hand there are almost 3 MB of free SPIFFS. 
So this problem must be solved in one or another way.

"If you flash such an image over an existing firmware you will loose your configuration (change in location of SPIFFS headers) and need to reconfigure the sensor."
The solution here might be to save configuration to the server and after upgrade read it from server.

> "With all those steps above we have too many steps that can go wrong."
That is why it is important to do it unit after unit. To update all sensors at the same time - is a dangerous approach. If something will go wrong - just email the owner to take some manual actions.

There might be other memory saving solutions:
1) To include only needed libraries like with languages: you select needed functionality (sensor, LCD screen or else) and your device is auto updated with the needed firmware.
2) Different versions of firmware: minimal, normal and extended - based on the hardware and the method of initial flashing.

dirkmueller, 2019-11-22:
This has been fixed using the two stage updater. we can now extend sketch size up to ~ 700kb (we need 300kb for the OTA updater). this should be enough. suggest to close.

closed 2019-11-27 by LuchtwachtersDelft

irukard, 2020-01-15:
Why didn't you saved config in binary format in emulated EPROM (4K at the end of flash chip), then reconfigure SPIFFS (all the data is lost because of new partition table) and restore config to new 2M SPIFFS?

dirkmueller, 2020-01-15:
@irukard because at the time this process was implemented that 2m/2m layout feature didn't exist. 

It was only introduced in the newer Arduino core version that we didn't use back then.

Using the two stage updater we can actually do this reformatting to the new 2m/2m format on the fly as well if we want to. So we have not given up flexibility..

There's actually a strong advantage  in the two stage logic: we could reflash the old firmware version from spiffs should the new version not be able to finish booting correctly since we have a backup if it available in spiffs.