opendata-stuttgart/sensors-software #328: NRZ-2018-115: DHCP-Request every minute!

dokape, 2018-11-20:
NRZ-2018-115
IP-Adress is assigned by a Raspberry Pi, Raspbian, by isc-dhcp-server.
lease-time is set to 12000 sec.

Checking the logfiles, I see the productive airrohr hammering the DHCP-Server with Requests for an IP-address.

With premier Version 111 I did not see this massive requests in the logfile.

Can anyone confirm this behavior?

````
Nov 20 08:50:14 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:50:14 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:40 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:40 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:40 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:40 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:41 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:52:41 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:53:03 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:55:48 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:56:10 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:58:55 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPDISCOVER from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPOFFER on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPREQUEST for 192.168.200.51 (192.168.200.2) from 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
Nov 20 08:59:17 pi3 dhcpd[10547]: DHCPACK on 192.168.200.51 to 52:c7:bf:1f:7b:b3 via enxb827eb0d7824
````

dokape, 2018-11-20:
Oh wait: 
Airrohr connected via repeater TP-Link RE500.
original MAC 60:01:94:1f:7b:b3 changed by repeater to 52:c7:bf:1f:7b:b3 (Prefix changed)
Seems to force airrohr request IP-address every minute.

ricki-z, 2018-11-22:
@dokape So is this an issue of the firmware or the used repeater? The repeater shouldn't change the MAC as it isn't requesting the lease.

dokape, 2018-11-23:
TPlink confirmed they change the first 3 Bytes of MAC for routing purposes.
I‘m still not shure what happened in case above. 
Sems to run now. 
I‘ll close this and keep on watching.

closed 2018-11-23 by dokape