opendata-stuttgart/sensors-software #330: Since firmware NRZ-2018-119A no data written to local InfluxDB

SmartBulb67, 2018-11-21:
Hi there,
I like your project very much and have already set up two sensors at different locations, which provide data since some months. They have the IDs 242418 and 4506372.
Both sensors also write data into different InfluxDBs.
One of the sensors (242418) stopped the data input into the InfluxDB exactly 2018-11-19 08:10 and one day later I realized the sensor has a new firmware (NRZ-2018-119A).

How could I check, if this firmware has something to do with the issue?
I could not find any debug messages in the web GUI.
Rebooting the sensor and re-setting the configuration I have already tried.

Any more ideas?

Best regards
Daniel

ricki-z, 2018-11-21:
@SmartBulb67 is moved.
And the answer: There was a problem with the latest firmware. We have set the update server to the previous version again, so a reboot should install this version and influxdb should work again.

SmartBulb67, 2018-11-21:
Hi @ricki-z ,
thank you very much both for the answer and for moving the issue.
I can confirm, the writing to the InfluxDB works again after the reboot.

Best regards,
Daniel

closed 2018-11-21 by SmartBulb67