opendata-stuttgart/sensors-software #334: Request: Support for Sensirion SHT31

OGVT, 2018-11-25:
Hi,

Currently I am testing the two sensors BHT22 and BME280. Especially in the current season we have a very high air humidity here. It occurs to me that with RH > 90% both sensors output very different values. The BHT22 saturates relatively fast and shows only 100%, while the BME280 hardly ever exceeds 90%. 

My Davis weather station uses a Sensirion SHT31. It seems to handle the high humidity much better. Will it be possible to operate a SHT31 (i2c) via the NodeMCU ESP8266 in the future?

CS, R

https://www.sensirion.com/en/environmental-sensors/humidity-sensors/digital-humidity-sensors-for-various-applications/

https://de.aliexpress.com/wholesale?catId=400103&initiative_id=AS_20181125114041&SearchText=sht31

ricki-z, 2019-11-06:
@OGVT should work with the actual release version. Could you please close this issue if it's okay?

OGVT, 2019-11-06:
Thank you very much!

OGVT, 2019-11-10:
Hi @ricki-z, can't see the SHT31 in the config. Which sensor type must I select in the configuration to address the SHT31?, BR, R

ricki-z, 2019-11-10:
Sorry thats on the beta branch, should be available when switching to beta versions or with next release. mixed up with other sensors added (noise, sps30)

OGVT, 2019-11-11:
Hi @ricki-z, can't find it in the latest_beta.bin. Should it be there?

ricki-z, 2019-11-11:
Which version number is shown on the sensor internal website?

OGVT, 2019-11-12:
Hi Rajko,

 

NRZ-2019-125-B1

 

 

 

 

Beste Grüsse

Roger

ricki-z, 2019-11-13:
This is the latest release version.
If you activate "use beta" on the sensor internal website the sensor should update to the latest beta.

OGVT, 2019-11-16:
works great!

 

Thank you.

 

BR, R

dirkmueller, 2019-11-27:
Thanks for the feedback!

closed 2019-11-27 by dirkmueller