opendata-stuttgart/sensors-software #335: PMS5003/7003 broken since Beta-Release (NRZ-2018-117?)

Adorfer, 2018-11-27:
On some Sensors PMS does not transmit any more the PMS. 
(no debug yet, if it's the reading from the sensor itself or just the data format of transmission to opensensemap. further investigation needed)

https://github.com/opendata-stuttgart/sensors-software/commit/49b8db5e11a7a6bde60701b123f6f5ce27b60b2e

NRZ-2018-117
https://opensensemap.org/explore/59eb9c3b49f6f80011ade240
https://opensensemap.org/explore/5b5a03bcd515090019e5adc5
[image]

lingling85, 2018-11-28:
the lastupdate keep all PMS Settings , After a reconfig it Works.

raivisdejus, 2018-12-16:
How does one access the config?
The 192.168.4.1 seems to have only config for WiFi, but how to enable the sensor to read PMS5003 instead of SDS011?

bertrik, 2018-12-16:
The initial screen only shows the items that you need to connect to the wifi plus the items to show debug info (like an OLED screen). Once the WiFi has been configured and rebooted, the web interface will show configuration items for the type of dust sensor, humidity sensor, etc. It will then no longer be at 192.168.4.1 but at the address assigned by your home router / access point.

ricki-z, 2019-02-14:
@Adorfer Can this issue be closed?

closed 2019-04-01 by ricki-z