opendata-stuttgart/sensors-software #336: TLS speed

ricki-z, 2018-11-28:
The latest update should also enabled TLS and certificate verification. But then the sensors needed too long to connect and to send data. The result was a large amount of open connections on the server. So I'm testing different ciphers and settings for TLS connections at the moment. This issues is for saving and publishing the results of these tests.
(Working version NRZ-2018-111: no cert verification, TLS1.1, AES256-SHA)

Connections without verification (all test are made with a cpu frequency of 80MHz):
default secure client: TLS1.2, AES256-SHA256 -> 1000ms

dirkmueller, 2019-11-27:
@ricki-z still an issue ?

ricki-z, 2019-11-27:
May become an issue again with BearSSL. It's something we have to have in mind when using SSL/TLS.
But it's not a real issue so I will close this. (Can look for this later)

closed 2019-11-27 by ricki-z