opendata-stuttgart/sensors-software #341: Timekeeping

LuchtwachtersDelft, 2018-12-12:
The 122 beta makes some changes to the NTP code. That got me thinking, how often is the time synced with NTP? Seems like acquireNetworkTime() is only done once at startup? 

The type of wireless modules we use drift one second to several minutes per day. I suggest syncing time a few times per day.

For related discussion and example code, see:
https://www.esp8266.com/viewtopic.php?f=5&t=10094

ricki-z, 2018-12-12:
The code in the actual beta is a first try. The needed refresh will be included later. We had to test the basic function first as most other NTP extensions crashed with our code. And it seems that some users connected the sensor to a guest wifi that also blocked the NTP port 123. So we may need some more precautions for these sensors for the functions needing an exact time (TLS cert verification, ...)

jordanc365, 2019-07-12:
You can set the sensors to restart sooner than the default 28 days and hence update the time. I have my sensors set to restart once a day and so far haven't had any problems :)

dirkmueller, 2019-11-27:
Actually the time is supposed to be synced once per hour automatically. I don't see an issue to be solved here?

closed 2019-11-27 by dirkmueller