opendata-stuttgart/sensors-software #342: DHT22 no values in luftdaten JSON

remyderuysscher, 2018-12-14:
The sensor shows DHT readings locally but not via luftdaten API.

Local
`{"software_version": "NRZ-2018-121C", "age":"12", "sensordatavalues":[{"value_type":"SDS_P1","value":"27.03"},{"value_type":"SDS_P2","value":"16.77"},{"value_type":"temperature","value":"4.90"},{"value_type":"humidity","value":"73.10"},{"value_type":"samples","value":"846782"},{"value_type":"min_micro","value":"162"},{"value_type":"max_micro","value":"26658"},{"value_type":"signal","value":"-87"}]}`

Luftdaten
`[{"sensor":{"pin":"1","id":7373,"sensor_type":{"name":"SDS011","manufacturer":"Nova Fitness","id":14}},"sensordatavalues":[{"value_type":"P1","id":5306090556,"value":"27.77"},{"value_type":"P2","id":5306090557,"value":"16.83"}],"timestamp":"2018-12-14 10:41:14","id":2495305266,"location":{"latitude":"52.0880","country":"NL","id":3727,"longitude":"5.0320","altitude":"0.2"},"sampling_rate":null},{"sensor":{"pin":"1","id":7373,"sensor_type":{"name":"SDS011","manufacturer":"Nova Fitness","id":14}},"sensordatavalues":[{"value_type":"P1","id":5306114617,"value":"26.70"},{"value_type":"P2","id":5306114618,"value":"16.87"}],"timestamp":"2018-12-14 10:43:42","id":2495316686,"location":{"latitude":"52.0880","country":"NL","id":3727,"longitude":"5.0320","altitude":"0.2"},"sampling_rate":null}]`

http://api.luftdaten.info/v1/sensor/7373/

What's up here?

ricki-z, 2018-12-14:
You have got 2 database IDs with the confirmation. ID 7373 is the ID for the SDS011 (PM). Die ID for the DHT22 is 7474 (in your case). So the correct link for the temperature/humidity data is http://api.luftdaten.info/v1/sensor/7373/

remyderuysscher, 2018-12-14:
Hello,

7474 is a different sensor located in bulgary?
I'm using home assistant integration with luftdaten this sensor expects 1
id with all sensor information. I see an empty value for temp. And Rh.

ricki-z, 2018-12-14:
Sorry right ID is 7374 and link is: http://api.luftdaten.info/v1/sensor/7374/
And for the Home Assistant integration you need to add multiple sensor instances. But please read the Home Assistent documentation for this.

remyderuysscher, 2018-12-14:
Allright, got it working. Was a bit confused because of the missing T en Rh
values. And the gui integration of home assistant allows only 1 sensor id.
Configuration in the YAML file works flawlessly.

Glad it's sorted out.

Thanks.

closed 2019-03-14 by ricki-z