opendata-stuttgart/sensors-software #345: NRZ-2018-121C: Can't save configuration

dr0i, 2018-12-22:
Cannot save configuration using Firmware: NRZ-2018-121C.
I want to share the data with opensensemap but the configs are not stored although I clicked on "save and reboot".
Maybe similar as https://github.com/opendata-stuttgart/sensors-software/issues/234.

rvowles, 2018-12-23:
I've newly arrived and flashed mine today, i presume same problem then.

ricki-z, 2018-12-23:
It's not similar to #234 (that was an error in a comparison).
Could you please check the newly uploaded release version NRZ-2018-123? This should be installed after a restart and was compiled again with the newer Arduino libraries. This libraries are using less RAM. It seems like the older library version (2.3) had a problem with our growing configuration if too many things (sensors and APIs) were activated.

rvowles, 2018-12-23:
I ended up downloading the code and upping the debug level, and found it hung at SPIFFs.begin - so instead built a hard coded version with my details and removed the dynamic config.