opendata-stuttgart/sensors-software #346: Sensor verbindet sich nicht mit meinem Wlan

Jadoch48, 2018-12-23:
Hallo in die Runde,
ich habe mir einen Feinstaubdaten-Sensor gebaut und schaffe es einfach nicht, ihn mit meinem Wlan zu verbinden.

- wenn ich mit dem Handy einen Hotspot erstelle und im Sensor die entsprechenden Login-Daten eingebe, verbindet das Gerät korrekt und die Messdaten sind unter https://www.madavi.de/sensor/graph.php?sensor=esp8266-3871409-sds011 einsehbar. Im Handy taucht er als "ESP_3B12B1" bei den verbundenen Geräten auf.

- wenn ich in der Sensor-Konfiguration die Daten meines heimischen Wlans eingebe, erscheint der Sensor nicht mehr als Hotspot und im Router-Menü erscheint der Eintrag "unknown" bei den verbundenen Geräten. Allerdings erscheint er auch nach längerem Warten nicht unter https://www.madavi.de/sensor/graph.php. Auch unter dem Link http://feinstaubsensor-3871409.local/ kann ich nicht auf ihn zugreifen.

- Ich habe ein 2,4Ghz und ein 5Ghz Netz mit selber SSID. Der Router ist ein  Arris TG3442DE (Vodafone vermarktet ihn als Vodafone Docsis 3.1).

Könnt ihr mir helfen? Ich weiß nicht, wie ich das Problem weiter eingrenzen kann.

Danke und viele Grüße
Jakob

ludwigm, 2019-02-13:
I have a similiar problem with a Vodafone EasyBox. I also tried to check with Wireshark and see a DHCP Discover package from the MAC address of the controller. It also shows up on the router but does not get an IP assigned. So I think it is mainly a DHCP issue. Would be great to have possibilities to set the IP instead of relying on DHCP for such cases. Not sure if it helps but I would hope. I added now a FritzBox to my Wifi and was able to connect through it but would for sure love to only need one access point (Vodafone EasyBox)

wertzui, 2019-02-22:
Same problem here:
ESP 8266 works with Fritz Box 6490, but not with TG3442DE.
But 2.4 networks are configured the same. All other devices connect fine to the new router without any changes (SSID and passwords are the same).
When I unplugged the TG3442DE and turned the Firts Box 6490 on again, the ESP 8266 was immediately able to connect to the Wifi network.

elmo-space, 2019-03-14:
I have the exact same problem with the TG3442DE router.
I analyzed the wifi traffic with Wireshark and it seems that the router is completely ignoring the DHCP discovers from the esp8266.
The ESP sends one DHCP discover after another and simply never gets a reply.

[image]

wetterfrosch, 2019-10-20:
Does the affected WLAN-accesspoint require some sort of "minimum basic-rate" which might be greater than 1 Megabits/s?

dirkmueller, 2020-01-04:
It is possible (I can't verify) that the wifi access point needs to support at least 802.11n. is it possible that the users with issues here are trying to connect to an AP without such support?

dirkmueller, 2020-01-04:
Google points to https://forum.vodafone.de/t5/Internet-Ger%C3%A4te/ESP8266-verbindet-sich-nicht-mit-TG3442DE/td-p/1974080 

not sure if the answer given here is correct though (I can still connect if I turn off on my 802.11n capable router the 802.11n feature)

elmo-space, 2020-01-05:
@dirkmueller It's most likely this. Unfortunately the Vodafone issued routers are built for idiots and don't offer *any* kind of settings to the user. You can't even change the wifi channel nor the dhcp settings in there. So theres probably not much that users of that router can do (other than setting up an external wifi router for the ESPs)

Stunkymonkey, 2020-02-04:
I think I was facing the same problem as you guys did, but now my esp8266 is working with my router:
```
Productname:               Vodafone Docsis 3.1
Firmware version:          01.02.037.03.12.EURO.SIP
```
the simple fix was:
update the core [esp8266 library](https://github.com/esp8266/Arduino/releases), rebuild and flash it

ricki-z, 2020-02-04:
If this should work then the next firmware release should solve this. We are waiting for the last feedback for some changes.