opendata-stuttgart/sensors-software #351: Difficult to tell apart 0 and 8 on SSD1306

LuchtwachtersDelft, 2018-12-29:
The characters 0 and 8 displayed on the SSD1306 are easily confused because in the used font the zero is a slashes zero. Because of that some people have a hard time connecting to the correct IP address or finding their sensor graphs in madavi.de. 

[image]

DigiE2, 2019-01-30:
Hi,
Here is the same...
Months ago i have tried to use a different font set.
I made very small changes (5 lines?) in version NRZ-2018-109. 
I used the font set from "OLEDDisplay.h" header "ArialMT_Plain_10" instead of "Roboto_Mono_9".
-> display.setFont(ArialMT_Plain_10);

The ArialMT_Plain_10 is not a monospaced font. 
Looke to the results - at the bottom are pictures with the new font set.
Number "0" and Char "M" are now much clear.

[image]
 and here
[image]

Wenn Interesse besteht und keine "Nebenwirkung" entstehen, könnte ich mich um die Änderungen auch in der aktuellen Version kümmern/committen.

Jens

ricki-z, 2019-02-26:
@DigiE2 wir würden gern bei einem monospaced Font bleiben. Das macht einfach das Einfügen neuer Texte etwas einfacher, weil man nur die Buchstaben zählen muss, um zu wissen ob der Platz reicht ;-)
Könntest du dir vorstellen, die Änderungen auch an dem Roboto-Zeichensatz durchzuführen? Das sollte ja prinzipiell ähnlich funktionieren.

DigiE2, 2019-03-29:
Wenn der Zeichensatz gleich bleiben soll, könnte man nur ein paar Pixel im Font ändern. z.B. in der Null.
Um die Lesbarkeit zu verbessern muss ein anderer Zeichensatz (Type,Höhe) her. Der Zeichensatz der in den 7x5 alphanumerischen Displays (AV-receiver/STB) genutzt wird ist Monospaced und besser lesbar. 
Beim SSD1306 passen damit mind. 16Zeiche in 6Zeilen.
Einen nicht mono Font finde ich trotzdem am besten, da auch mehr Zeichen dargestellt werden können.
Die den seltenen Fällen von zu langem Text sollte der Displaytreiber dann des Ende begrenzen bzw. abschneiden.

ricki-z, 2019-04-01:
Gerade bei den langen Zeilen wäre es ungeschickt, wenn etwas abgeschnitten werden müsste. Das betrifft z.B. den Namen des Sensors. Wenn dann die letzten Ziffern fehlen ...
Oder die IP-Adresse, die ist max. 4*3 Ziffern + 3 Punkte lang = 15 Stellen lang. Aber mit einem Nicht-Monospaced Font hängt es dann an den einzelnen Ziffern wie lang das wird. Und auch hier ist das Fehlen der letzten Stelle(n) eher ungünstig.
Wir haben hier leider scheinbar niemanden, der einen Font für den Treiber editieren kann. Daher wird es wohl erst mal der aktuelle Font bleiben.

LuchtwachtersDelft, 2019-04-19:
The Roboto Mono 9 used here doesn't look like a monospaced font either. Small m has 5 pixels and capital M has 4 pixels. Has the font been mangled somehow?

gerald-b, 2019-05-13:
Ich finde auch, das 0 und 8 schlecht zu Unterscheiden sind.
kann der strich in der 0 nicht einfach entfernt werden?

Bei dieser Anwendung sollte doch eine verwechslung mit 0 (null) und O (Otto) fast ausgeschlossen sein.

ricki-z, 2019-05-13:
The firmware source code (including the font sources) is available here. So if someone can change this ...

Ted-CAcert, 2020-03-29:
Is this still relevant? I just installed an Adafruit SSD1306 OLED with the default "latest" firmware, and I don't have slashed zeros...