opendata-stuttgart/sensors-software #354: No Temperature metrics send to influxdb

oschni, 2019-01-02:
Hello,
I also send data to my local influxdb.  The temperature data from the BMP180 seems to be missing.

FW; NRZ-2018-123B

```
> select * from feinstaub
name: feinstaub
time                SDS_P1 SDS_P2 max_micro min_micro node            samples signal
----                ------ ------ --------- --------- ----            ------- ------
1546428246261162930 20.4   5.6    27396     72        esp8266-7222354 1951702 -66
1546428412294292763 19.7   5.7    142619    74        esp8266-7222354 1811606 -67
1546428561158032539 21.87  5.33   132285    78        esp8266-7222354 1777904 -67
...
```

ricki-z, 2019-01-02:
You should chekc the configuration of the sensor. It's seems like the BMP180 isn't activated.

oschni, 2019-01-02:
I checked the configuration, the BMP180 was activated. So i checked the page "Aktuelle Werte" and see no temperature anymore.

 Together with "Send to influxdb" I also activated "Auto Update". I think that updated the sensor to the latest FW NRZ-2018-123B and stopped working the BMP180. For the last few days it shows the temperature on the page "Aktuelle Werte".

ricki-z, 2019-01-03:
Is the data for the BMP180 shown on the page "Aktuelle Werte" at the moment?
Maybe a complete power cylce can solve this. Sometimes the BMP and BME sensors aren't initiated correctly and doesn't send data. And you should check the cables to the BMP180.

ricki-z, 2019-11-06:
@oschni is this problem solved?

dirkmueller, 2019-11-27:
@oschni  Please reopen when the issue isn't solved. Assuming it was a setup error for now. Thanks for reporting!

closed 2019-11-27 by dirkmueller