opendata-stuttgart/sensors-software #356: Lokale Daten ohne Wifi

erniberni, 2019-01-05:
Wäre es möglich die Software so anzupassen, dass man den Sensor dauerhaft als Accesspoint betreibt und darüber die Daten "Aktuelle Werte" zumindest lokal anschauen kann. Ich sehe das als Anwending für einen mobilen Sensor, der nicht grundsätzlich ein WLAN haben wird.

jmberg, 2019-01-06:
Wenn ich den Code richtig verstehe, sollte folgendes gehen: Setze "Dauer Routermodus" auf einen sehr grossen Wert (maximal wohl 2147483, gerunded 2^31/1000), dann musst du (bei dem genannten Wert) nur alle ca. 3 1/2 Wochen eine HTML-Seite aufrufen damit der Sensor quasi immer in "lokalem AP" Modus bleibt.

ricki-z, 2019-01-07:
Setting "Duration router mode" doesn't work. Whil in this mode the sensor won't make any measurements.
Adding a display is the prefered solution. Or you can disable the APIs (so the sensor won't try to send data to them) and connect the sensor to a smartphone or mobile hotspot. As you need a device to show the values this should be equal to the solution with the sensor as an hotspot.