opendata-stuttgart/sensors-software #359: Daten an Homematic Variable senden.

capitaenz, 2019-01-08:
Hallo zusammen,

ist es möglich die Daten über "An eigene API senden" an eine Homematic Systemvariable zu senden?
Ich habe es wie unten im Bild zusehen ist ausprobiert, aber es klappt so leider nicht.

[image]

VG
Leif

ricki-z, 2019-01-08:
Das wird so einfach nicht gehen. Wir senden bei "An eigene API senden" ein JSON-Objekt an die angegebene Adresse. Ein paar weitere Infos dazu findest du im Sourcecode und unter https://github.com/opendata-stuttgart/meta/wiki/APIs

capitaenz, 2019-01-09:
Hab ich mir schon fast gedacht... schade.
Beim Projekt EasyESP gibt es die Möglichkeit verschiedene Protokolle auszuwählen, mit "Generic HTTP" kann man somit Daten an eine Systemvariable unter Homematic beschreiben.
Ist so eine Möglichkeit vielleicht mal für die Zukunft angedacht?

ricki-z, 2019-01-10:
Das ist nicht wirklich angedacht. Bis auf wenige "Spezialfälle" ist dies eigentlich nicht nötig, da fast alle Systeme mit JSON umgehen können.

capitaenz, 2019-01-11:
Könnte mir vorstellen das noch mal ein ganzer Schwung neuer Sensoren zum Netzwerk dazukommen, wenn es eine "Homematic Unterstützung" geben würde  ;-)
Ich habe jetzt allerdings den Sensor mit einem im HM-Forum veröffentlichten Script einbinden können.
[https://homematic-forum.de/forum/viewtopic.php?f=18&t=43201&hilit=Luftdaten.info&start=20](url)
Besser ist allerdings immer aus Sicht der HM-Zentrale, wenn die Daten von extern geliefert werden, anstatt diese holen zu müssen.

VG
Leif

ricki-z, 2019-01-11:
Wir haben halt nur begrenzten Speicher auf der NodeMCU zur Verfügung. Vor allem der RAM spielt eine Rolle. Daher müssen wir uns schon überlegen, was wir alles umsetzen. Da etwas generisches vom Nutzer definieren zu lassen, dürfte schon einige Bytes belegen. Dazu kommt dann noch das "Zusammenbauen" der entsprechenden Daten ...
Und es gibt ja eine Möglichkeit, die Daten in die Homematic zu bekommen. Alternativ zu dem Script kann die Homematic ja inzwischen auch direkt als "Empfänger" programmiert werden. Beispiele dafür findet man unter https://github.com/hobbyquaker/awesome-homematic/blob/master/README.md#interfacing-software . Wenn sich also jemand findet, der ein "Virtual Interface" für die Homematic schreibt, könnte das dann das JSON Objekt verarbeiten.
Habe aber bitte Verständnis, wenn das aus der Homematic-Community kommen muss und nicht auch noch von uns umgesetzt werden kann.

andreas-bulling, 2019-04-08:
I wondered if somebody has a similar solution for HomeAssistant? #405