opendata-stuttgart/sensors-software #364: NRZ-2018-123: changing sending_intervall_ms

RRM1, 2019-01-12:
Hi,

I would like to modify the value of sending_intervall_ms to e.g. 4 minutes. 
The (default) value defined in airrohr-firmware is 145000 (-> 2.5 minutes).
I would like to change it to 4 minutes. 
Modified code:
     // unsigned long sending_intervall_ms = 145000;
     unsigned long sending_intervall_ms = 235000;
But after flashing the modified SW the measuring interval is still 2.5 minutes.

KR
RRM1

RRM1, 2019-01-12:
I can change the value via the Web interface of the sensor (configuration -> measurement interval 145 -> 235.

ropeters68, 2019-01-12:
If you want to reduce data transfer you can also switch-off the https option. This lowers the data to less than 4 Mb per day ( standard (574 measurements/day)

ricki-z, 2019-01-12:
You need to use a fresh NodeMCU or you must delete an existing configuration first. Otherwise the existing config value of 145 seconds (created by the previous firmware version) will be used.