opendata-stuttgart/sensors-software #365: Delete test data from API

michalskop, 2019-01-14:
How can I delete some test data already sent to the API / shown on chart at madavi.de?

The idea is to have only "real" data as public (and not the data from the test phases).

bertrik, 2019-01-15:
I have no idea, I don't know of any way to do this. But I think it is probably just way too much work to handle requests such as this.

ropeters68, 2019-01-15:
Deleting is not an option but maybe a simple solution is to add a test flag to the confugation screen and add the flag to the message so test data can be identified afterwards?

michalskop, 2019-01-16:
Just to add some context - my usecase is that I am testing (learning) the stack and therefore I can have very different measures (at home, etc.), not real outside measures.

ricki-z, 2019-01-23:
We use the RRDTools for the Madavi.de graphs. It's not possible to change the data of the round robin data files directly. The only way is dump the data, then delete the test data and play back the rest.
Even if we would implement this (hoping, that not too many people need this), we would need to implement a user management and a registration.

michalskop, 2019-01-23:
I understand
Maybe than to think about it as  "how to implement some test regime"?
The thing is that default settings is "plug in and start sending data to madavi and luftdaten" - which makes sense, because it is simple. On the other hand, the first data will be probably usually from indoor, testing? (At least it seems to me.)
But may be the history is not so important comparing to the real-time data (?)

dirkmueller, 2019-12-02:
We can add a 'test mode' button to ui that doesn't get saved in config, so will disable itself after reboot and stops sending data.

Is that what you're looking for?