opendata-stuttgart/sensors-software #366: Not able to configure custom API (anymore)?

michalskop, 2019-01-14:
Is it (still) possible to config local or custom API for the data?

The is just this page with no options for that (and "Back to home page" returns to the same page)

[image]

ropeters68, 2019-01-14:
The screenshot looks like the initial page to set-up the Wi-Fi connection. After save/restart you can logon again using the IP-address provided by your router, then you'll get the full configutation screen (including the API-option).

michalskop, 2019-01-15:
Thanx @ropeters68!
I did not get it from the manual.

For future reference:
The settings of wifi (the first time or after a change of wifi/router) is at 192.168.4.1 (it is available when the board cannot connect to the internet).

However, the settings of the sensors, APIs, etc. is at the address given by the router if it is connected, e.g. 192.168.1.11 (one can get the address from the router admin pages, which may be at 192.168.1.1)

closed 2019-01-15 by michalskop