opendata-stuttgart/sensors-software #367: Wifi Repeater or Ap cONNECTION

gaga9999, 2019-01-16:
Hallo,
Ist es möglich aus mehreren gleichen SSID zu wählen, zB nach Mac Adresse? Ich habe abbrechende Wifi Verbindungen und denke der ESP verbindet immer mit dem Repeater, statt der Basisstation, in deren Nähe er ist?

ricki-z, 2019-01-23:
Hallo,
bei einer gleichen SSID ist es nicht möglich, das vom Sensor gewählte Gerät zu ändern. Die Software sollte automatisch den stärksten verfügbaren Zugangspunkt wählen.
Über welches Gerät der Sensor an das Netz angebunden ist, sollte sich aber über die entsprechenden Seiten auf den Geräten herausbekommen lassen. Bei Fritzboxen ist dies in den aktuellen Versionen auf "Heimnetz -> Mesh" zu sehen.
Zu beachten ist, das die NodeMCU nur auf 2,4 GHz arbeitet. Wenn also nicht alle APs auf 2,4 GHz senden, können die entsprechenden APs auch nicht genutzt werden.

dokape, 2019-04-09:
Alternative: für den Sensor ein eigenes z.b. "Gastnetz" am gewünschten Zugangspunkt (hier Router) einrichten. Ich habe das so am Repeater gemacht.

tobox, 2019-08-20:
Ich habe ca. 20 ESP8266 im WLAN laufen, alle mit Tasmota Firmware. Ich habe eine Fritz!Box mit 3 Repeatern mit LAN-Uplink. Wenn die Repeater geupdatet werden verbinden sich alle Tasmota-Geräte nach einigen Minuten wieder mit dem stärksten Access Point. Nur der Feinstaubsensor klebt dann dauerhaft an einem sehr weit entfernten Access Point fest, und verbindet sich nicht mit dem stärksten. Es wäre schön, wenn die über SetOption56 und SetOption57 aktivierbaren Optionen zum Roaming in Tasmota in die Feinstaubfirmware übernommen werden könnten.