opendata-stuttgart/sensors-software #368: Config page doesn't work in Microsoft Edge

LuchtwachtersDelft, 2019-01-23:
After typing the IP address of the sensor in Microsoft Edge nothing happens.

Workaround: use another browser like Firefox or Chrome.

ricki-z, 2019-01-23:
Or wait some seconds and try again ...
I have checked this with an actual version of the Edge browser. The page is shown there.
We have to stop the internal webserver of the sensor while the data is sent to the APIs. While this the page can't be opened (in all browsers).

LuchtwachtersDelft, 2019-01-23:
Believe me, I waited and refreshed.

On 2 separate computers, typing the IP addresses of 2 different sensors, in 2 different houses, by 2 different people, in Microsoft Edge has the same result: nothing happens.

Right now I can put Edge and Firefox next to each other on the same screen, type the same IP and only Firefox shows the page.

[image]

ricki-z, 2019-01-29:
Checked with both Edge and Firefox. Page is shown in both modes (private and public).
Private mode should allow more than public mode (i.e. showing printers and shares). So that is not logical why you should be able to open the page in a public network but not in a private network.

LuchtwachtersDelft, 2019-01-29:
Exactly, it's not logical. It might be Microsoft bug, but it's still a real problem for some Luftdaten users whose default browser is Edge. I don't usually use Edge, so I didn't notice it before. This happened during a workshop on a participant's laptop and I reproduced it at home.

When I searched for this Windows 10 + Edge + not loading pages problem I found quite some results, such as: 

https://techcommunity.microsoft.com/t5/Windows-10/Windows-10-1809-Edge-not-loading-pages/td-p/266050

https://answers.microsoft.com/en-us/edge/forum/all/ms-edge-not-opening-private-websites-when-using-ip/bdd552e9-6deb-46e6-aa4c-f100ad72c967

Whether or not this is something out of our control, we should at least acknowledge it as a known issue.

ricki-z, 2019-01-29:
As far as I can read in these articles it is a problem with IPv6 disabled. This should be the case only in some cases . By default Windows has enabled IPv6 since minimum version 7.

LuchtwachtersDelft, 2019-02-18:
I just read the release notes of the 2019-02 Cumulative Update for Windows 10 Version 1809:
 
> The update fixes a bug in Microsoft Edge that prevents the browser from connecting to a site using an IP address. 

Could that be it? Going to back-up, restart and test.

LuchtwachtersDelft, 2019-02-18:
I confirm the config page loads in Edge after applying the the 2019-02 Cumulative Update.

closed 2019-02-18 by LuchtwachtersDelft