opendata-stuttgart/sensors-software #370: Parameters in config page not properly setup in new sensor.

RikDrabs, 2019-01-24:
We have assembled & tested more than 1000 sensors here in Brussels. On 4 or 5 we saw the following:
During setup of SSID & password to the WiFi network, the firmware autoupdate was not selected.
On subsequent testing with the sensor, it would not transmit data to Luftdaten & Madavi. The sensor would show "Sensor nicht gefunden" on the graphics display, even after hours of waiting time.
The reason is simple: when opening the complete configuration page all options are off. 
API luftdaten & API madavi are off, explaining the lack of communications; SDS011 & DHT22 are off, so probably no measurements are made, and firmware autoupdate is also off.
Can this situation not be prevented? It is not normal that all those options are off. If this could be detected & corrected during startup, then field problems could be avoided.

ricki-z, 2019-01-25:
With which firmware version were the sensors flashed?
Normally (with autoupdate enabled by default) the APIs Luftdaten.info and Madavi.de and the sensors SDS011 and DHT22 are activated. This is also saved in the configuration when saving the wifi credentials.

ricki-z, 2019-01-25:
This should only disable auto (OTA) updates. But you mentioned this for the 4 to 5 problematic sensors. This could have limited the source of the problem.
But if this was only in older versions we may have "solved" the problem.

RikDrabs, 2019-01-25:
OK. But maybe there is another cause ...
The link between autoupdate off, and sensors and APIs off was clear. Maybe there was, or is a common cause that turns everything off. Anyway, if we see the problem again during final testing, with recent firmware (121C or 123B), we will notify again.

ricki-z, 2019-03-12:
@RikDrabs can we close this issue by now?

RikDrabs, 2019-03-12:
Yes this issue can be closed. We didn't see any other problems in newer
sensors ( with 121c or 123b)

closed 2019-03-13 by ricki-z