opendata-stuttgart/sensors-software #372: unit constantly reconnecting to the WLAN

patrislav1, 2019-01-25:
Hi,

my airrohr unit is constantly reconnecting every 2-3 minutes to the WLAN. (I found out when I discovered logs with lots of entries on my FritzBox). It loses connection and then reconnects again immediately.
See excerpt from the router's log below.

The signal displayed on the airrohr's WebUI looks ok (-68 dbM, 64%)
is this a known issue?
do I have to assign a particular WLAN channel?

Thanks

```
25.01.19 18:12:55 WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
25.01.19 18:12:55 WLAN-Gerät hat sich abgemeldet (2,4 GHz), Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
25.01.19 18:10:15 WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
25.01.19 18:10:15 WLAN-Gerät hat sich abgemeldet (2,4 GHz), Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
25.01.19 18:07:37 WLAN-Gerät angemeldet (2,4 GHz), 72 Mbit/s, Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
25.01.19 18:07:37 WLAN-Gerät hat sich abgemeldet (2,4 GHz), Feinstaubsensor, IP 192.168.111.28, MAC 2C:3A:E8:08:DD:3F.
```

ricki-z, 2019-01-25:
Is the Fritzbox configured to hide the SSID (name of the access point)?

ricki-z, 2019-01-25:
Which firmware version is installed?
I also have a Fritzbox and my sensor is running since more than 5500 measurement cycles. Firmwareversion is NRZ-2018-123B, connected are a SDS011 and a BME280. APIs used: Luftdaten.info, Madavi.de, Opensensemap.

patrislav1, 2019-01-25:
my unit runs the same version, it has a SDS011 and a DHT22. it's using the Luftdaten.info and Madavi.de APIs.

ricki-z, 2019-02-01:
The timespan is more or less the measurement interval. So it seems like the sensor tries to send data and can't do this. Or can you see any data in the graphs on Madavi.de?
Are you using any other API (InfluxDB, Eigene API, Opensensemap)?

dirkmueller, 2019-11-27:
@patrislav1 is this still an issue with current firmware?

dirkmueller, 2020-01-04:
Please reopen when information can be provided. Thanks!

closed 2020-01-04 by dirkmueller