opendata-stuttgart/sensors-software #374: Reset SDS011 on startup

kwaaak, 2019-02-01:
If the SDS011 sensor has been used before, and the client was not careful, it might not be in the right state to work with the Luftdaten.info-firmware.

The solution would be to always reset the sensor during startup. This library describes the issue:
https://gitlab.com/frankrich/sds011_particle_sensor/blob/master/README.md#get-started

and has a function to perform the reset:
https://gitlab.com/frankrich/sds011_particle_sensor/blob/master/Code/sds011.py#L544

ricki-z, 2019-02-01:
If you look at the code we do this "reset" (there is no real reset command for the SDS011, you can only set the default values). We set all sensors to the continuous reading mode which is the default.

kwaaak, 2019-02-01:
I looked through the code:
https://github.com/opendata-stuttgart/sensors-software/blob/6369abaeace3f28bcca7e7b3c7bb17f65c0effc4/airrohr-firmware/airrohr-firmware.ino#L611-L622

It seems to be missing the reset of data reporting mode, page 2 of the protocol pdf:
https://cdn.sparkfun.com/assets/parts/1/2/2/7/5/Laser_Dust_Sensor_Control_Protocol_V1.3.pdf

This should be it:
`0xAA, 0xB4, 0x02, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0x01, 0xAB`

My C++ is not robust to make a pull request, someone else should take this responsibility :innocent:

ricki-z, 2019-02-02:
We won't add this. We need to save every single byte. If someone is thinking he needs to set parameters then he should be able to reset this.

dirkmueller, 2019-11-22:
This has been fixed as part of #379 imho. agree to close or is there anything left to be done?

dirkmueller, 2019-11-27:
Please don't hesitate to reopen if there's still anything left. I reviewed that we're setting all settings that are persistent on power on testing now.

closed 2019-11-27 by dirkmueller