opendata-stuttgart/sensors-software #377: Not clear that auto update must be on to change the language.

DeeKey, 2019-02-06:
Language change requires to download a new firmware.
This is only possible with "auto update" option ON.
It is not clear from UI.
Can the following note be added to the language change feature: "Turn auto update ON!"

https://github.com/opendata-stuttgart/sensors-software/issues/254

dirkmueller, 2019-11-04:
The alternative is that we do an "autoupdate" if the language has been changed from the currently configured ones. that seems to be more seamless imho (but it will change the version to the latest one)

DeeKey, 2019-12-30:
Lets close this issue! 
Problem solved!

dirkmueller, 2019-12-30:
Nothing has been done yet, I still want to make this more obvious as many are falling into this trap.

dirkmueller, 2020-01-04:
Has been fixed in -B7 now, closing.

closed 2020-01-04 by dirkmueller

DeeKey, 2020-01-04:
Yes - now it can be close as it is absolutely clear that autoupdate is required to change the language!

commit 58a00616 (2020-01-04): Disable editing of Beta/language when OTA is turned off (Fixes #377)

This adds a small javascript snippet to toggle the
editability of the other checkboxes.