opendata-stuttgart/sensors-software #378: simple API for integrating in existing project?

michapr, 2019-02-09:
Hi,
is there a simple way to send the ready values to your database?
Have running the SPS30 here local (integrated in my own solution) but would be ready to share the results.

Is there a simple API call to send the values in any documented format?
As sample a HTTPS call with API key inside (like for some weather stations)?

Thanks for any hint.

ricki-z, 2019-02-12:
Please look at the documentation at https://github.com/opendata-stuttgart/meta/wiki/APIs . There you can find examples of API requests. For a new sensor we need to define a new 'type' and the export format (which fields should be saved).
We are also working on the support for the SPS30, so we may add this sensor to our list of supported sensors in the near future.

stewecar, 2019-02-12:
The sensor claims an autoclean funcion and a long life cycle..maybe due to a filter mounted in the main hole.
It would be nice, if possible, to launch a collective purchase cause the cost is three four time of a pms 7003 or the glorious sds011..

michapr, 2019-02-12:
@ricki-z If I understand right I could use the "POST for SDS011" with the right parameters ?
It is not depending on the type of sensor, right?

@stewecar I think a "collective purchase" maybe not very effective because of transport costs to every end point ;)  It is not so expensive - you can get it from 26.50€ + VAT in Germany (as sample) - without connector & cable

stewecar, 2019-02-12:
> @stewecar I think a "collective purchase" maybe not very effective because of transport costs to every end point ;) It is not so expensive - you can get it from 26.50€ + VAT in Germany (as sample) - without connector & cable
I didn't know that the price is about 30-31 euros..here in Italy is about 10/12 euros more..so i have written that is not cheap at all..but surely affordable..also if all luftdaten project i think today is about only 22-23 euros (wifi module, sds011, cables, etc..).
About the "collective purchase" you're right..in fact i think that it make sense only if the order is about hundreds of pieces, where the discount could be at least about 10 euros per piece..

michapr, 2019-02-25:
@ricki-z - can you please check it (using SPS30 data)?
> If I understand right I could use the "POST for SDS011" with the right parameters ?
It is not depending on the type of sensor, right?

And - is it possible to import older data to the system - or does it not make sense for you?
(like I made it at OpenSenseMap)
Thanks!

ricki-z, 2019-02-26:
At the moment we don't add older data to the system.
For the SPS30 you could use the same parameters as we use for the PMSx003 sensors. Only the prefix of the value names needs to be changed.

ropeters68, 2019-02-27:
@ricki-z - do you have some more info on when the support of the SPS30 will be ready?

ricki-z, 2019-02-28:
It's not as easy as for all the other PM sensors. The SPS30 could be connected via I2C or UART. But for the UART interface we need to set a speed of 115200 baud. This is a little too fast for the NodeMCU, even if SoftSerial should support this. But as our software may also need some cpu cycles, some of the packets (~5%) can't be read.
And for I2C you need a slightly different cabling. So 2 resistors (pull-up) needs to be added and pin 4 needs to be connected to GND. So this kind of connection isn't as easy as those of the other PM sensors.
At the moment we are testing the UART interface. If the lost data is distributed equally we can use this. But if we can't get enough samples per measurement interval we may have a problem ...

dirkmueller, 2019-11-27:
Sps30 support has been added in the last stable firmware. Also there is the possibility to post data to the API directly using other code as well. Closing as I see nothing left to be done here. Please reopen if I am missing something. Thanks!

closed 2019-11-27 by dirkmueller