opendata-stuttgart/sensors-software #379: If SDS011 is put in query reporting mode no data is collected

philippejadin, 2019-02-10:
I built a sensor using an SDS011 used in another project. The project involved using this library : https://github.com/lewapek/sds-dust-sensors-arduino-library
The sensor was put in query mode instead of active reporting for his particular project.

If the sensor is put in query mode with `sds.setQueryReportingMode(); `the luftdaten code doesnt work anymore.

This is written to the sensor so it is kept after power off.

Solution : set the sensor to active reporting mode on startup. Although I don't know how to do it precisely, some idea can be taken from here : https://github.com/lewapek/sds-dust-sensors-arduino-library/blob/master/src/SdsDustSensorCommands.h#L105 

I hope that helps those who might think they have a dead sensor.

niceguydave, 2019-07-03:
This is really useful, thanks.  I'm in a situation like this now.  I can hear that the SDS011 sensor is working and can hear it firing into action at the beginning of every measurement process but no data ever makes it through my ESP-12E Micro USB ESP8266 CP2102.  My thoughts up to this point is that there was some kind of data loss through the connections but this should help me try and diagnose things better.

niceguydave, 2019-07-03:
For what it's worth, @philippejadin, the following project has some really useful ways of interacting with the sensor: https://gitlab.com/frankrich/sds011_particle_sensor

ferbar, 2019-07-12:
1. There seems to be a bug in the SoftwareSerial library. If you connect the GPIO4 of the esp to a serial - usb converter an connect it to a pc the starting 0xAA of a command is mostly recognized as 0xA8. So sending commands might work ... or not ...
Maybe upgrading the SoftwareSerial will fix that. I'll give it a try.
2. The SDS011 code is a mess, command replys are not read, so nobody sees if the command is not received. I'll write a PR

ferbar, 2019-07-15:
@1: If I add a delay(100) before sending bytes with the SoftwareSerial libraray chances are good that the bytes are transmitted successfully.
Next try: switching to ESPSoftwareSerial (https://github.com/plerup/espsoftwareserial). This worked! However I was using esp8266 board 2.4.0 and the library was written for 2.5.x, so some lines have to be changed for 2.4.0. This is not an option for everyone...
Next try: upgrading esp8266 board to 2.5.2 and using the builtin SoftwareSerial library. Bytes are transmitted successfully in most cases. Wowever: the web interface is broken and the esp8266 resets with "soft wdt reset"

My search ended at these 2 lines (...)
https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/airrohr-firmware.ino#L3746
https://github.com/esp8266/Arduino/blob/2.4.0/cores/esp8266/Esp.h#L87

ferbar, 2019-07-15:
dup #374

dirkmueller, 2019-11-11:
@philippejadin this has been fixed as part of #374. can this be closed?

philippejadin, 2019-11-12:
@dirkmueller thank you for the follow up and thanks to the community for the fix!

commit e670ed2d (2019-11-12): Set SDS011 to active reporting mode when doing power on test (Fixes #379)

It is a really painful error to debug when it the sensor
accidentally was switching into explicit reporting mode, and
those few bytes don't hurt. Adding the delay also recovers a
weird failure of my SDS after reset.
closed 2019-11-12 by philippejadin