opendata-stuttgart/sensors-software #380: Arduino 1.9.0-beta - error: default argument given for parameter 4 of 'String form_checkbox(const String&, const String&, bool, bool)' [-fpermissive]

imaluschi, 2019-02-11:
Hi,
when trying to compile the airrohr-firmware with Arduino 1.9.0-beta an error occurs:
`error: default argument given for parameter 4 of 'String form_checkbox(const String&, const String&, bool, bool)' [-fpermissive]`

When examining the problem i saw the definition of `form_checkbox` awaits 4 variables, but the call have only 3 values.
`String form_checkbox(const String& name, const String& info, const bool checked, const bool linebreak = true)`
f.eks: `form_checkbox(name, add_sensor_type(info), checked);`

This is obviously no problem for Arduino 1.8.8, but on 1.9.0-beta it doesn't work anymore.

I'm not a programmer, so I can't say if this would be anything to change/repair.
To solve the problem:
add `, false` of `, true` tho the calls of `form_checkbox`.
`form_checkbox(name, add_sensor_type(info), checked, false);`

Used the version NRZ-2018-124-B1, but i think it's the same for previous versions.
Nevertheless, add at least an Information that Android up to 1.8.8 have to be used (and not 1.9.0-beta).

ricki-z, 2019-02-12:
Could you please check the actual beta branch? It seems like Arduino changed the behaviour of 'default' vars in functions calls.

imaluschi, 2019-02-12:
Hi,
arduino-PR-beta1.9-BUILD-25, downloaded from arduino.cc
Maybe I'll try a recent version from Github.

ricki-z, 2019-02-12:
I meant those of our firmware, not the Arduino IDE.
And normally we check our software source code only against release versions of the Arduinio IDE as there may be failures and (feature) changes in later beta releases

imaluschi, 2019-02-12:
Sorry, I misunderstood :-/
Tried the Version NRZ-2018-124-B2 and it works as expected.

Thank you for your work!!

closed 2019-02-12 by imaluschi