opendata-stuttgart/sensors-software #383: Acces Point mode: wifi password transmitted unencrypted? Maybe a security problem

speendo, 2019-02-19:
A friend and me, we are thinking about building a Feinstaubsensor, therefore I had a short look at the project and the code.

In the first setup, when you have to enter the settings for your home wifi, is this information transmitted encrypted?

On first sight it seems to me that it is not.

In this [line](https://github.com/opendata-stuttgart/sensors-software/blob/7becd7643740820514bb6d64ab9f167685a165a6/airrohr-firmware/airrohr-firmware.ino#L2033) the AccessPoint is established with a password stored in `cfg::fs_pwd` however, as this [line](https://github.com/opendata-stuttgart/sensors-software/blob/7becd7643740820514bb6d64ab9f167685a165a6/airrohr-firmware/ext_def.h#L15) shows, `cfg::fs_pwd` is an empty string.

Therefore, it seems to me that an attacker that "listens" to the communication between me and the ESP8266 while entering the settings for my home wifi could get access to my network like this.

Maybe I missed some other security measures you have taken. If not, an easy solution would be to set `cfg::fs_pwd` to something else than an empty string.

I've also noticed this issue: https://github.com/opendata-stuttgart/sensors-software/issues/128 however, I couldn't see if this particular problem was already solved with the measures you were taking there.

ricki-z, 2019-02-19:
The password is transmitted unencrypted on send.
What can we do?
- using HTTPS: this would need a certificate, but self-signed certs are throwing a warning (confusing most users)
- using WPA2: we need to set a random password. But how should we tell this password to the user? Not all users are able to read the console output. If you think you will have to change the password regularly, then you can set an own password for the sensor in AP mode

Normally the wifi password should be transmitted only once unencrypted at the moment of saving (and only if changed). In all cases where a 'password' is shown in the input fields only a string with asterisks is transmitted. So someone needs to capture the traffic in exactly this timespan where you configure the sensor the first time.

If you want to avoid the the unencrypted transmission you could compile and flash a modified version of the sensor software where you can predefine your wifi password.

mdoersam, 2019-03-19:
Just setup two sensor-systems and realized this issues exists right after. I totally agree with speendo. Just "Password" as a password would be a big improvement as it prevents sniffing of the actual wifi-password while it is being transmitted.

This still leaves the issue that someone could be able to reconfigure the sensor, but I don't really see the security-issue with that, as it runs with default configuration at that point

speendo, 2019-04-01:
I've made up a pull request to overcome this issue. Only a minor changes. However, the manual(s) would have to be adapted...

Please consider accepting the pull request as this problem could possible cause some harm.

mkg20001, 2023-09-30:
Another idea would be to use OWE for the AP which does not require a PSK

bertrik, 2023-11-19:
IMO, you should always consider what is exactly the problem.
Is it that, with the WiFi password, anyone can access the internet over your access point?
Or is it that other people can snoop on YOUR network if they have the WiFi password?

speendo, 2023-11-19:
@bertrik I think the question is valid in general.

However, in the particular case I would say that the exact problem is, that your wifi password can not be considered secret, when you transfer it on an unencrypted connection.

A potential attacker could do several bad things once they have your wifi password INCLUDING but not LIMITED to accessing your internet connection and snooping your network.

As the fix of this problem is not difficult and I reported this issue three years ago, I think it would be about time to implement a solution, regardless of what the worst kind of attack would look like.