opendata-stuttgart/sensors-software #385: Support for mobile data transfer GSM / GPRS

wparuch, 2019-02-23:
Is there any chance to extend the project with packet data transmission? It would be very useful in places where there is no wifi, and the air is polluted :). For example, using the SIM800 / SIM900 / SIM908 / A6 / SIM5320 / SIM5360 / SIM7100 modules.

ricki-z, 2019-02-26:
This should be possible. But we dont have such a module to implement and test this.

samtech7, 2019-05-28:
Could the A6 quad band Module https://www.androegg.de/?product=ai-thinker-a6-gprs-gsm-quad-band-modul be a solution? I think combined with 1once.com it could be quite cost effective.

ropeters68, 2019-05-28:
That would be nice! In one case of bad WiFi I use a separate MiFi 4G router and a prepaid data sim card.  Please note that if the https option is active the data usage is much higher (x3?). If switched off 1 Gb of data should be enough for at least 9 months (with normal cycle and madavi/luftdaten/feinstaub app options active) of data.

MagTun, 2019-07-23:
Is there a way to connect  the sensor to a smarphone which has Wifi Hotspot enable as well as data enable (3G/4G)?

ricki-z, 2019-07-24:
In this case you can connect the sensor the same way as with any other wifi router.

MagTun, 2019-07-24:
Thanks ricki-z! It's working great. I have created a [page](https://github.com/opendata-stuttgart/meta/wiki/How-to-use-SDS011-as-a-mobile-PM-sensor) in the wiki that explains all the steps:

ricki-z, 2019-07-26:
Thanks for creating the wiki page. There is only one addition from my side. Mobile sensors should be marked as "Indoor" at my.luftdaten.info if the sensor is configured to send the data to Luftdaten.info.
Otherwise the "static" location info might be used by people that want to work with our data. Indoor sensors should normally be filtered out by them.

MagTun, 2019-07-26:
Ok, I just added this. 
Maybe a next improvement would be to add a support for mobile sensor (with GPS data from a GPS module).

ricki-z, 2019-08-02:
This is a planned feature. But we need to make some changes to our API and database for this. At the moment our database can't use any location data sent together with the measurements.