opendata-stuttgart/sensors-software #386: NRZ-2018-123B: BME280 keine Daten mehr nach x Stunden

brandungskieker, 2019-02-24:
Verwende das Standard-Setup 
SDS011
BME280
mit Firmware NRZ-2018-123B

Vielleicht hat jemand einen Rat: In unregelmäßigen Abständen stoppt der BME280 Sensor bei mir die Übermittlung der Daten. Teilweise werden auch verkehrte Daten geschrieben. 
Im "Aktuelle Werte" Screen erscheint ein "-" bei den Werten. Ein Hard- oder Softreset hilft, danach werden wieder Daten gesendet. Die Regelmäßigkeit der "Sensor hang-ups" sieht man gut auf den Sensorgraphen:
https://www.madavi.de/sensor/graph.php?sensor=esp8266-830217-bme280

[image]

ghoeffner, 2019-02-25:
Evtl. related zu #384? :)

muesli, 2019-03-16:
Kann ich hier auch feststellen.

ricki-z, 2019-03-16:
@brandungskieker aktuell scheint der BME280 keine Aussetzer oder falschen Werte zu haben. Sehe ich das richtig? Wurde etwas geändert? Wäre interessant für die Leute mit dem gleichen Problem.

brandungskieker, 2019-03-18:
@ricki-z es hat sich wieder beruhigt, habe am Sensor nichts verändert. Die Software ist meines Wissens die gleiche geblieben, oder gab es jüngst Updates?

ricki-z, 2019-03-18:
Das letzte Update war im Dezember, die meisten Sensoren sollten das auch zeitnah erhalten haben.

dokape, 2019-04-09:
Mit dem BME280 und der FW -123b läuft mein Sensor seit Wochen problemlos.
Vielleicht ein temporäres Kontaktproblem? Feuchtigkeit? Wackelkontakt? feine Erschütterungen?

ricki-z, 2019-06-05:
@brandungskieker können wir den Issue schließen?

brandungskieker, 2019-06-09:
Nicht auszuschließen dass es auch mit einem Defekt des Sensors zusammenhängt.

closed 2019-06-09 by brandungskieker