opendata-stuttgart/sensors-software #388: Measurement floor pm10 at ca 30 -  sensor might need cleaning

gaga9999, 2019-02-28:
Hy, my sensor started to constantly show at least 30 mikrograms at pm10, pm2.5 was ok (sensor id 21052). today i blew into the hose and now the measurements are ok again. you can see the drop in the graphs. so some dust was stuck in there.. 

i seem to see the same problem elsewhere, i.e. see sensor id 4963 - high constant pm10 but not so high 2.5.

maybe a "offset" detector might be useful in the software?

ricki-z, 2019-03-01:
An "offset" detector implemented directly in the firmware is a problem. We are even now short below the memory limit. So https connections are causing crashes sometimes because of not enough RAM.
But we are working on a server based solution where we can access data over a longer timespan to detect such anomalies. The user will then get an email notification with hints how to solve the problem.

ghoeffner, 2019-08-28:
This seems to be a problem with quiet a lot of sensors, especially with the project aging. For us blowing into the sensor only worked for a couple of weeks before the problem returned and the ultimate solution was just to replace the sensor. There was not too much visible dust in the old sensor, so it's probably an aging problem. 
[image]
[image]
(Thanks to Josef for these pictures)

dirkmueller, 2019-12-02:
@gaga9999 let me see if I understand this correctly. 

The firmware should automatically determine a baseline, e.g. if there was never measurement below 30 (arbitrary example), then it should just subtract that minimum from all measurements?

I think this is a dangerous approach, because nothing says that the systemic error is linear.. assuming this is caused by dust on the camera lens, those pixels are simply 'blind' and will not see anything, including dust/no dust. Which means we not only have an offset but also a higher uncertainty what the correct value is.

Imho there is no way around fixing the sensor, which might require un assembling and cleaning it

dirkmueller, 2019-12-19:
Closing due to lack of replies.

closed 2019-12-19 by dirkmueller