opendata-stuttgart/sensors-software #389: How to rotate SSD1306 Display 180°

Obicom, 2019-03-06:
Hi, first of all thank you for the awesome Luftdaten project. I am new in this "business" and made my first station with SSD1306 OLED display and BME280 sensor in addition. My SSD1306 have a yellow part for the title line but with the default configuration the title line is on bottom (°°°°) dots. Is it possible to rotate the display 180° without new compiling of the firmware?  If not what I should change in the firmware to rotate the display? Thank you in advance for your kind support.

Obicom, 2019-03-06:
Solved it with a new compilation and an additional command ... but maybe as a feature request for a new version .. would be great if it would be implemented in the WebUI as an option.

[image]

ricki-z, 2019-11-06:
@Obicom can we close this issue? This option should be available in the latest release version.

Obicom, 2019-11-06:
@ricki-z Yes, you can ...

closed 2019-11-06 by ricki-z