opendata-stuttgart/sensors-software #390: SDS011 doesn't sleep/fan doesn't stop

vb0, 2019-03-07:
My sensor doesn't seem to sleep (at least the fan doesn't stop). It's returning periodically (plausible) values (so bidirectional communication with the sensor works I guess and the sensor is somehow sane), it stays connected to network/internet, the web interface is accessible.  
I THINK it used to stop some months ago when I put this together the first time (but didn't use it for a long time). Is there some setting somewhere I need to watch out for? Also how to debug this, I have two generic questions:

- you can set the debug level but where is it going? I presume over it's own serial but no way to get it to log some other place too (rsyslog or similar)?
- can you change the firmare OTA to some specific version (not the automatic updates)?

=====================
Nova PM Sensor SDS011
Firmware: NRZ-2018-123B
Measuring interval: 145000 (the default?)

ricki-z, 2019-03-07:
The debug output can only be viewed via the serial interface. Adding features like rsyslog or similar that only may be used by a handful of users would make the resulting firmware image significantly bigger. This space we would like to use for more useful features (or new sensors). At the moment we use 92 percent of the available flash memory (only 512kByte, otherwise we couldn't use OTA).
OTA to specific version is also a feature for few users. We would need to save all versions, the code for the update server needs to be changed. Why not flashing the specific version?

vb0, 2019-03-07:
Well it is pretty clear that there are benefits for a networked device (especially one that is meant to be mounted outside who knows where) to be actually able to upgrade the firmware and do debugging over the network (assuming of course it is in a sane enough state) without taking it of the pole and connecting it to a serial programmer or similar. Tasmota can do it very well on similar hardware (as far as the ESP goes) and it can do a pretty decent amount of processing, sensor management and so on. Sure, if this is where we are it's fine but the answer to "why" should be pretty obvious.

Back to sleeping - anything in particular that needs to be set on the SDS011 itself or some known bug in my version (I assumed it might be version related as I didn't change anything else, but who knows, this is why I wanted a quick way to flash it to a previous one)?

YaguraStation, 2019-03-07:
@vb0 This might not be useful for your particular case, one thing that causes the fan of an SDS011 to continually run is wrong wiring (I think TX and RX swapped).

vb0, 2019-03-07:
Ha, thanks I actually just figured it out! It is wiring! Not TX/RX as it was getting data but a I had a floating ground (I disconnected the +5V and the GND from the sensor to log it in another power supply -I want to characterize the power consumption on the 5V rail- and everything was floating so I had no signal ground between the ESP and the SDS. Surprisingly enough it was still limping like that and managed to communicate even without ground but when the ESP wasn't sending anything it was like it was not connected/high impedance and that if I remember right makes the sensor up.
Good, thank you again.

closed 2019-03-07 by vb0