opendata-stuttgart/sensors-software #391: BME 280 in addition ?

bianco57, 2019-03-10:
Hello the latest version of the software (NRZ-2018-123B) allows to add a BME280 on a current configuration (SDS011 and DHT22) ? 
bme280  => connecting refer to version 3 of the NodeMCU ? 

Thnx

YaguraStation, 2019-03-10:
Please find info herehttps://github.com/opendata-stuttgart/meta/wiki/EN-Assembly-of-components-(circuit)

ricki-z, 2019-03-10:
You can find a description of how to connect the BME280 on this page:
https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#bmp180--bmp280--bme280-i2c

bianco57, 2019-03-15:
hello thanks for the answers I just instalelr a bme280 in the same device with a dht22 ... if the TH is the same for the two sensors by not against the HR !?? does the bme tend to underestimate the hygrometric value ? my professional weather station (DAVIS Pro) gives me a 95% "reference"value ...
http://valdefench57sgz.franceserv.com/pws/index.php?lang=en
[image]

thank you and have a good weekend Regards

DigiE2, 2019-03-29:
Maybe you has the same problem like this - (german):
https://github.com/opendata-stuttgart/sensors-software/issues/170#issuecomment-346306126

I notice that the humidity values of the DHT22 are over 70% too high.
I use for this  https://opensensemap.org/explore
Compare the humidity of neighboring stations with different sensors (mostly applies - is only TH, HM -> DHT22 - or has TH, HM, Pressure -> BME280 or other no DHT).

I tend to trust the BME values.

dokape, 2019-04-09:
I can confirm this behavior. 
The humidity of the DHT22 is often different.

FrankUlbrich, 2019-05-31:
Hallo,
ich baue das Projekt gerade in zweifacher Ausführung nach und habe mehrere Feuchtesensoren im Wohnzimmer verglichen. Nachdem ich mich an zwei Referenzsensoren orientiert habe, fällt mir auf, daß meine BME 280 Sesnoren ca. 10-15% Punkte zu wenig Feuchte anzeigen.
Ich fände es sinnvoll, einen änderbaren Offset vorzusehen. Ich habe dies in meinen Sourcecode einprogrammiert. Dennoch fände ich es handlicher, wenn sich der Wert im Webinterface unter"Konfiguration" parametrieren lassen könnt. Das möchte ich vorschlagen.
MfG
Frank

ricki-z, 2019-06-01:
Die Firmware ist eh schon sehr knapp mit Speicher. Die Konfigurationsseite können wir im Accesspoint-Modus schon gar nicht mehr komplett anzeigen. Deshalb möchten wir auf solche "Spezialkonfigurationen" in der allgemeinen Firmware verzichten.
Ist der BME280 relativ weit oben im Gehäuse angebracht? Dann kann schon ein Verschieben Richtung Öffnung das Problem verkleinern.

FrankUlbrich, 2019-06-03:
Ich habe eine Testinstallation mit einem Wemos D1 mini pro (16MB) durchgeführt. Die Sensoren befinden sich auf einem Testboard nebeneinander ohne Gehäuse unter meinem Hausdach. Von dem Wemos hatte ich noch ein paar Modelle liegen. Das Schönen an der neuen Version ist, daß dieser mehr Speicher und einen Anschluß für eine externe Antenne besitzt und kleiner ist.
Ich nutze für meine Testversion dann eine angepaßte Firmware mit einem Offset für den BME280

Danke für die Rückmeldung
Frank

krantic, 2019-10-16:
> You can find a description of how to connect the BME280 on this page:
> https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#bmp180--bmp280--bme280-i2c
BME280 => Pin 11
Where can I find Pin 11?
Or is it D0 (https://github.com/opendata-stuttgart/meta/wiki/Pinouts-NodeMCU-v2,-v3) ?

ricki-z, 2019-10-16:
Could you please read the README.md completely? What you have found is the X-Pin header value for the Luftdatn.info API.
If you click on the link mentioned in your comment https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#bmp180--bmp280--bme280-i2c you will find the folowing:

    VCC -> Pin 3V3
    GND -> Pin GND
    SCL -> Pin D4 (GPIO2)
    SDA -> Pin D3 (GPIO0)

dirkmueller, 2019-11-28:
Description on how to connect bme280 had been provided.

Please open a separate issue is the humidity precision problem ist still there with current firmware. Thanks!

closed 2019-11-28 by dirkmueller