opendata-stuttgart/sensors-software #392: Any plans to make custom API and add HTTPS ?

mercteil, 2019-03-11:
Hello. I like the project quite a bit and use it to develop some custom API to receive the sensor data and push it to my influxdb with a dashboard.

However it seems it doesn't push to my server (not local) and also does not support HTTPS.

Is there any way you will add HTTPS and custom API support outside the local network?

Best regards... stuggi ftw

ricki-z, 2019-03-11:
For the custom API you can use any server, even a server outside a local network. And if you set the port to 443 HTTPS will be used.
BUT: your server has to support POST requests and should not use redirects.
So if your 'non-local' server doesn't accept the data please look at the logfiles.

dimisus, 2019-03-11:
Yep, that worked, thanks. I was just going through the docs and read it was meant to be used within your local network.

ricki-z, 2019-03-11:
Where have you found this info in our documentation? I think we should change this ;-)

dimisus, 2019-03-11:
https://github.com/opendata-stuttgart/meta/wiki/APIs#an-eigene-api-senden

ricki-z, 2019-03-11:
Ich habe die Seite geändert. Passt das so?

dimisus, 2019-03-12:
Top.

ricki-z, 2019-03-12:
So we can close this issue?

closed 2019-03-12 by ricki-z