opendata-stuttgart/sensors-software #393: Lokale JSON API

muesli, 2019-03-16:
Bietet die Firmware auch eine lokale JSON API an?

ricki-z, 2019-03-16:
Unter http://<ip-addresse-des-sensors-in lokalen-netz>/data.json können die Daten abgerufen werden. Zu beachten ist, das wir den lokalen Webserver während des Sendens an die Server deaktivieren, da sonst der Speicher knapp wird. Die Datei ist also alle 2,5 Minuten kurzzeitig nicht erreichbar.
Falls der Issue damit erledigt ist, bitte schließen.

muesli, 2019-03-16:
Das wirft mir nur ein 404 not found zurueck.

ricki-z, 2019-03-16:
Sorry github hat einen Teil der Adresse "geschluck". Dort sollte stehen http://{ip-adresse-des-sensors-im-lokalen-netz}/data.json

muesli, 2019-03-16:
Keine Sorge, hatte ich schon richtig verstanden :-) Dennoch: 404 not found.

ricki-z, 2019-03-16:
Was komisch ist, weil es bei meinem Sensor mit der aktuellen Firmware funktioniert.
Was passiert bei http://{ip-adresse-des-sensors-im-lokalen-netz}/ ?

muesli, 2019-03-16:
Ich erhalte das Admin/Config Interface, um meine Adresse & Geodaten sowie Sensortyp und ID des SDS011 einzustellen.

ricki-z, 2019-03-16:
Dann läuft auf diesem Sensor NICHT unsere Firmware. Hier ist also das falsche Repo für deine Frage. Bitte wende dich an die Entwickler der Firmware, die auf deinem Gerät läuft.

muesli, 2019-03-16:
Huch?! Vielen Dank fuer die Info.

closed 2019-03-16 by muesli

ricki-z, 2019-03-16:
Interessant wäre aber, welche Firmware da läuft. Bzw. von wem die ist?

muesli, 2019-03-16:
Das frage ich mich nun auch. Die Firmware wurde beim smartair Workshop in Augsburg zur Verfuegung gestellt. Mir wurde gesagt, es handelt sich um dieses Repo hier, genauer gesagt:

https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware

ricki-z, 2019-03-16:
Wir fragen aber weder die Adresse oder Geo-Daten ab noch die ID des SDS011 (diese können wir über die Software auslesen). Von daher kann es einfach nicht unsere Firmware sein.
Kannst du mir einen Screenshot schicken (rajko(at)codefor.de)?

muesli, 2019-03-16:
Klar, kein Problem:

[image]

Firmware Binary stammt scheinbar von hier: http://www.teco.edu/research/mitmachen-im-smart-air-quality-network-workshop-augsburg/ (CrowdSensor v0.8)

Ich hoffe du kannst etwas Licht ins Dunkel bringen, ich bin endgueltig verwirrt :confused:

ricki-z, 2019-03-16:
Würde mich ja mal interessieren, wiele Leute bei dem Workshop waren und wieviele davon eigentlich einen Sensor für Luftdaten.info bauen wollten.
Denn der dort gebaute Sensor sendet vielleicht auch an uns, aber ohne Registrierung ist der auf unserer Karte nicht zu sehen. Und zum anderen sendet er an einen Server am KIT. Die dort ankommenden Daten werden aber aktuell gar nicht veröffentlicht. Das macht dann X Bausätze, die still und leise vor sich hin messen. Das Projekt des KIT findest du unter http://www.smartaq.net/STARTSEITE/

muesli, 2019-03-16:
@ricki-z Danke, das klaert auch gleich eine weitere Frage, die ich hatte: wieso ich mich auf luftdaten.info separat registrieren muss :wink:

Mal sehn, vielleicht tausche ich dann einfach selbst die Firmware.

Am Workshop haben ca. 30 Leute teilgenommen. Ich moechte betonen, dass es auch gut moeglich ist, dass ich schlichtweg etwas falsch verstanden habe. Die schriftliche Anleitung spricht tatsaechlich auch nur von einem "Sensor nach dem Vorbild von luftdaten.info", verlinkt aber auch dorthin und zu euren Seiten.

ricki-z, 2019-03-16:
Andere Firmware wird schwierig, weil da ein ESP32 eingesetzt wird. Aber vielleicht bekommen wir das auch so hin. Siehst du auf der Seite irgend eine Art Seriennummer (außer der des SDS011)?

muesli, 2019-03-17:
Spielst du jetzt auf die separate luftdaten.info Registrierung an? Wenn ja, dann hab ich das bereits gemacht, warte allerdings immer noch darauf, dass mein Sensor hinzugefuegt wird. Ich kann dir die Sensor ID aber gern nochmal per Mail schicken!

ricki-z, 2019-03-17:
Ja, schicke mir die Sensor ID bitte noch einmal per Mail. Ich sehe hier leider EMail-Adresse nicht und habe dadurch nichts, wonach ich suchen könnte.