opendata-stuttgart/sensors-software #394: [BME280] Completely wrong values for temperature and pressure all of a sudden

PattaFeuFeu, 2019-03-17:
Vor wenigen Tagen sind die Werte meines BME280 auf absurde Weise ins Falsche abgedriftet. Von einer Minute auf die nächste sank der Temperaturwert auf konstante -144,3°C, bei denen er seitdem auch geblieben ist.

Am Aufbau habe ich in den letzten Monaten seit Ersetzen des DHT22 durch eben jenen BME280 nichts geändert. Da ich Auto-Update eingeschaltet habe weiß ich nicht, ob jetzt gerade eben erst auf die Version NRZ-2018-123B aktualisiert wurde oder ob diese Version schon länger installiert war.

Ich weiß auch nicht, ob es sich hierbei um einen Hardwaredefekt oder ein Softwareproblem handelt. Die Luftdruckwerte fluktuieren noch – aber dennoch rund 300 Einheiten unter dem, was umliegende Wetterstationen messen –, der Temperaturwert bleibt konstant, der Luftfeuchtewert scheint halbwegs akkurat, wenngleich dieser zum Zeitpunkt des Absinkens der anderen Werte auch recht sichtbar gesunken ist.

[image]

[image]
[image]
[image]

Weiß jemand Rat oder hat dasselbe Problem?

ricki-z, 2019-03-17:
Einen Softwarefehler würde ich ausschließen, aktuell arbeiten über 1400 Sensoren mit BME280 mit der aktuellen Firmwareversion. Und weniger als 1% bringen offensichtlich falsche Werte.
Fehlerquellen können sein:
schlechte Steckverbindungen
"kalte" Lötstellen
Probleme mit der Spannungsversorgung
auf den BME280 gerichtete Lichtquellen (warum auch immer Licht ein Problem ist...)

dokape, 2019-04-09:
FW -123b hat bei mir keine Probleme mit dem BME280.
Schlechte Steckverbindung, Kalte Lötstellen, Stromversorgung, defekter BME280.

kplaczek, 2019-04-10:
I had that too couple weeks ago with my BME280 connected to the raspberry pi in my case. Checked that sensor with another raspberry and my ESP8266. The same values everywhere. The values like this mean the sensor is busted and only thing you can do is replace it or return that for a new one to the seller. Sorry i don't speak German, and this answer is based just on a question.

PattaFeuFeu, 2019-04-10:
I have redone the soldering and tested the BME280 isolated on a different Pi without the Lufdaten software and it still provides the same read outs. I have thus ordered a new BME.

closed 2019-04-10 by PattaFeuFeu