opendata-stuttgart/sensors-software #395: BM280 air pressure value stored wrong

ituri, 2019-03-17:
I'm using a BME280 connected to Feinstaub-sensor and sending the data to a local InfluxDB instance. However, it turns out that the air pressure values are stored in a very strange and wrong format. Additionally, sometimes (I can't exactly reproduce when that is), an additional number gets added to the air pressure value.
In order to show you the problem, here's a dump of my table (please note that I've anonymized my sensor ID):
`> select * from feinstaub
name: feinstaub
time                BME280_humidity BME280_pressure BME280_temperature SDS_P1 SDS_P2 max_micro min_micro node          samples signal
----                --------------- --------------- ------------------ ------ ------ --------- --------- ----          ------- ------
1552836943584624256 78.36           100048.66       6.72               1.77   1.43   931141    75        esp8266-31337 1798162 -73
1552837092089435493 74.64           100036.44       6.8                1.3    1.2    932328    77        esp8266-31337 1796507 -73
1552837240425117750 72.82           100024.45       7.04               1.7    1.4    930751    76        esp8266-31337 1804440 -72
1552837388761806510 73.25           100032.59       7.3                1.4    1.3    932443    78        esp8266-31337 1751277 -73
1552837537368080572 73.53           100037.73       7.4                1.77   1.2    930646    77        esp8266-31337 1775085 -73
1552837685704251263 70.45           100044.36       7.22               2.23   1.3    932329    78        esp8266-31337 1765004 -73
1552837834040668861 71.85           100043.81       7.11               2.57   1.2    930236    76        esp8266-31337 1822905 -73
1552837982445691200 73.29           100047.94       7.13               1.8    1.27   932873    76        esp8266-31337 1812562 -72
1552838130951308198 71.71           100054.32       7.15               2.07   1.1    930552    76        esp8266-31337 1813307 -72
1552838279357516414 71.73           100057.72       7.07               1.7    1.12   932848    76        esp8266-31337 1808698 -73
1552838427832276929 69.88           100057.33       6.98               1.57   1.23   932321    76        esp8266-31337 1806673 -73
1552838576184871747 70.16           100052.31       7.05               1.35   1.25   932338    77        esp8266-31337 1776692 -74
1552838724600140282 70.86           100055.24       7.06               3.37   1.3    932754    77        esp8266-31337 1770093 -74
1552838872981838047 71.12           100061.53       7.13               2.98   1.1    932256    79        esp8266-31337 1741791 -72
1552839022127262844 70.17           100072.67       7.15               1.87   1.2    932511    77        esp8266-31337 1778192 -74
1552839170532982501 72.29           100071.83       7.1                1.55   1.45   932372    78        esp8266-31337 1770497 -73
1552839318941197458 71.08           100073.16       7.14               1.37   1.2    932336    78        esp8266-31337 1781332 -74
1552839467345755692 70.13           100073.08       7.28               3.08   1.52   931129    78        esp8266-31337 1742121 -73
1552839620910661390 70.52           100075.56       7.29               2.23   1.67   932387    77        esp8266-31337 1791630 -71
1552839769249924477 69.89           100086.32       7.22               3.95   1.72   943095    77        esp8266-31337 1774371 -71
1552839920350473586 71.18           100094.84       7.2                2.27   1.47   932262    79        esp8266-31337 1729042 -72
1552840068688135055 69.47           100099.14       7.25               2.17   1.1    932556    77        esp8266-31337 1762058 -71
1552840218832713885 69.51           100096          7.26               1.87   1.3    932415    79        esp8266-31337 1746720 -71
1552840367239766942 70.4            100092.73       7.24               1.45   1.35   932804    79        esp8266-31337 1731633 -70
1552840515576710405 71.09           100107.06       7.33               1.57   1.47   932452    78        esp8266-31337 1775601 -70
1552840663913884585 70.9            100112.47       7.36               4.63   2.83   932358    79        esp8266-31337 1734890 -71
1552840814948709751 69.59           100105.81       7.45               2.85   1.75   932476    78        esp8266-31337 1774909 -67
1552840963377261085 69.73           100108.26       7.47               2.58   1.47   932556    79        esp8266-31337 1729626 -71
1552841115873032378 69.79           100120.66       7.33               4.23   1.6    932241    79        esp8266-31337 1741758 -71
1552841264210953251 69.88           100119.19       7.38               2.9    1.5    932417    78        esp8266-31337 1754043 -71
1552841417662034352 70.78           100123.38       7.23               1.23   1.2    932428    78        esp8266-31337 1765687 -71
1552841565999941034 69.97           100128.11       7.2                3.43   1.6    932652    79        esp8266-31337 1742885 -71
1552841714504460483 69.77           100125.63       7.18               3.07   1.6    931794    78        esp8266-31337 1759346 -71
1552841863383916346 69.56           100138.6        7.2                1.93   1.4    932487    78        esp8266-31337 1775361 -71
1552842021747469122 67.87           100127.77       7.19               1.87   1.33   931993    77        esp8266-31337 1779648 -71
1552842170422728524 67.81           100148.06       7.13               1.4    1.3    931882    77        esp8266-31337 1770044 -70
1552842319567385833 66.4            100136.66       7                  1.3    1.2    933618    78        esp8266-31337 1771923 -71
1552842467905977238 65.73           100134.44       6.87               1.92   1.37   931764    77        esp8266-31337 1776847 -70
1552842629593952098 66.83           100138.16       6.85               1.32   1.15   931628    79        esp8266-31337 1744455 -70
1552842778100667868 66.24           100146.23       6.87               2.12   1.17   931734    77        esp8266-31337 1774557 -70
1552842930401644067 65.61           100148.75       6.75               2.07   1.1    932332    75        esp8266-31337 1788053 -70
1552843078725888794 67.64           100149.2        6.71               1.2    1.1    931495    76        esp8266-31337 1773945 -70
1552843232370499525 65.84           100144.78       6.71               3.4    1.1    931201    77        esp8266-31337 1787436 -70
1552843380778583900 66.87           100149.2        6.71               1.72   1.4    932154    77        esp8266-31337 1773003 -70
1552843529112763569 66.22           100158.19       6.68               1.83   1.1    929152    76        esp8266-31337 1782045 -70
1552843677449417928 65.2            100168.3        6.63               1.42   1.32   931879    78        esp8266-31337 1758176 -70
1552843825955248003 65.96           100161.91       6.66               2.5    1.03   931807    78        esp8266-31337 1758722 -71
1552843974561940714 66.08           100173.84       6.6                1.92   1.1    931932    78        esp8266-31337 1749100 -70
1552844123036386726 65.83           100170.56       6.62               2.77   1.1    931684    79        esp8266-31337 1750776 -70
1552844271374404228 65.33           100174.12       6.75               2.5    1.15   931102    77        esp8266-31337 1754120 -70
1552844424932450426 65.43           100179.23       6.76               1.6    1.2    931155    80        esp8266-31337 1754881 -70
1552844573338313224 64.91           100181.16       6.92               3.37   0.95   931120    79        esp8266-31337 1740496 -70
1552844725284105328 64.34           100194.13       6.76               1.73   1      931574    78        esp8266-31337 1772144 -70
1552844873688089964 64.23           100200.91       6.69               2.02   1      931713    79        esp8266-31337 1733900 -70
1552845023535423179 65.03           100219.86       6.63               1.47   1.27   931233    78        esp8266-31337 1757128 -70
1552845171869539297 64.99           100222.97       6.62               1.98   1.1    934115    78        esp8266-31337 1752902 -70
1552845320206520657 65.61           100222.78       6.6                4.37   1.13   931252    77        esp8266-31337 1805312 -70
1552845468713502959 66.15           100230.35       6.49               3      1.12   931186    77        esp8266-31337 1789769 -70
1552845619729622794 66.89           100228.52       6.45               1.2    1.1    931641    77        esp8266-31337 1801143 -70
[... cut ...]
As you can clearly see, the decimal separator is located at the wrong position of the value. Additionally, the first values somehow have an additional zero inserted to them (c.f. `100044.36`).
At the same time, I can assure that the web interface of my Feinstaub sensor always showed the correct values, so I suppose that this problem occurs at the point where the firmware tries to store the data in the database. 
Also, the values on madavi.de were always displayed and rendered correctly. If they had been transmitted with the additional zero in it, I would have seen a spike in the graphs, which I haven't.

ricki-z, 2019-03-17:
pressure values are sent in Pascal, not hectoPascal. This is the unit read from the sensor. And as we save raw data the pressure is sent in Pascal.
Any modification/conversion of data is done on the raw data (i.e. on the web interface or Madavi.de)

ituri, 2019-03-17:
Okay, that seems understandable. However, it still leaves me with the problem of the strange numbers at the beginning of my records where the additional zero is inserted. Any explanations for that?

ricki-z, 2019-03-17:
I can't see an example in the list?

ituri, 2019-03-17:
Just look at the first 24 values, starting from the top. For example the first air pressure value is `100048.66`. Even if you store them in Pascal, the value should be `10048.66` with one less 0 in it.

ricki-z, 2019-03-17:
1 hecto Pascal is 100 Pascal, not 10 (this would be deci).

ituri, 2019-03-17:
Sorry, I just figured out my fallacy. 
1. Simply divide by 100, I realised that after your first hint.
2. I looked at the series of data and recognised a pattern in it which lead me to believe that the values stored in the database were wrong. However, the air pressure here only fluctuated by about 3 hpA throughout the above records. So everything's fine on your end. :)

Thank you very much for the quick help!

closed 2019-03-17 by ituri