opendata-stuttgart/sensors-software #396: more efficient display on LCD1602

poinck, 2019-03-20:
In my opinion, when in normal operation, only sensor data should be displayed (all sensors in one screen: *pm10*, *p2.5*, *temp*, *humidity*). Only when starting the ESP8266 things like FW version, ESSID name, chip ID and IP should be displayed. The **sensor screen** could look like this (including an indicator until next measurement will occur: `seconds / 10 = c`):

```
p10=200  p25=100
t=-23 h=100 c=99
```

- [ ] omit fractions of sensor data, because they give a false sense of accuracy, anyway
- [ ] remove units to save space
- [ ] better/optional: make display screen configurable in configuration web interface

https://github.com/opendata-stuttgart/sensors-software/blob/6369abaeace3f28bcca7e7b3c7bb17f65c0effc4/airrohr-firmware/airrohr-firmware.ino#L3437

ricki-z, 2019-03-20:
Sometimes you need to know the chipID or the IP to access the sensor config page. As we use DHCP the IP address may change from time to time. So its useful to see this information on the display. Otherwise you have to restart the sensor.
And some sensor have a pressure sensor installed (the values there have a range from 950 to 1100).
PM10 values can reach more than 1000 to New Year so you need also one character more there.
Your version was the first e implemented and was changed on request of many users.

ricki-z, 2019-03-20:
Addition:
Some of the displays are used to show the values to people not knowing our device. So you need the units.
And additional config options need space in flash and on the config page. The config page must fit into memory and there are only some kByte left. So an option used only by a few users may block the addition of a future option we really need.

poinck, 2019-03-20:
Ok, I understand. So, i will only create a patch for myself to be close to the upstream version. And thanks for the hint about the value range:

```
p10=1000   t=-23
p25=1000   h=100
```

closed 2019-03-20 by poinck