opendata-stuttgart/sensors-software #398: Proxy config in AP Mode

rohit-rcrohit7, 2019-04-01:
Hi,

My question is very specific to Luftdaten project. It uses a Nodemcu which starts in station mode (AP) to connect to an available WiFi. But the WiFi here in question is behind a firewall and needs prxy server and port to be provided for authentication. My Nodemcu connects to the WiFi and gets an IP address but as expected can't send the the data iver to the madavi server. Is there a way to configure proxy details? ( It's a school WiFi where am trying to monitor air pollution)

Thanks!

ricki-z, 2019-04-02:
In the actual version it's not possible to configure a proxy. And it's also not planned. We are running out of available flash memory. So we can't implement all features.
May be it's possible to set the sensor on something like an exception list?

rohit-rcrohit7, 2019-04-02:
Ahh I imagine, 4MB is not a lot to work with. You are right, quiite possibly, that would be the way out. ( If we manage to track down the IT guy to do so).  Thanks!

closed 2019-04-28 by ricki-z