opendata-stuttgart/sensors-software #404: No sensor readings from SDS011

andreas-bulling, 2019-04-07:
Hi,

I've just put together an air quality sensor from source (Firmware: NRZ-2018-123B). Platform is a WEMOS D1 ESP8266.

DHT11 works fine but I can't get any sensor readings from the SDS011. I've connected it as described (RX, TX), the fan is running, and this [Python script](https://github.com/aqicn/sds-sensor-reader/blob/master/sds-reader.py) also shows values for PM2.5 and PM10 when connecting the sensor directly to the PC via USB.

However, I am not able to get any sensor readings with the WEMOS D1. Debug Info High doesn't show anything useful.

Any ideas/suggestions for how to debug this?

DmitrSou, 2019-04-07:
Stupid suggestion - do you mark the sensor in config?

ricki-z, 2019-04-07:
Is it possible that the sensor was reconfigured before with another software? The sensor needs to be set to continuous mode, not passive mode. And some test scripts change this to passive more.

andreas-bulling, 2019-04-07:
> Is it possible that the sensor was reconfigured before with another software? The sensor needs to be set to continuous mode, not passive mode. And some test scripts change this to passive more.

Possible but unlikely - I only plugged it into the USB port or connected it to the D1 so far. The simple Python script does no configuration at all - it's just a simple serial connection, in contrast to the firmware - not sure why..

andreas-bulling, 2019-04-07:
> Stupid suggestion - do you mark the sensor in config?

yes

> If so - I would double check RX/TX connection...

I did - including swapping the cables... :(

DmitrSou, 2019-04-08:
Is there possible to read diagnistic output via serial? If so I would change log level to Debug and see what happens. Actually I've never have deal with WEMOS D1 device...

andreas-bulling, 2019-04-08:
I tried many combinations and finally found a working setup:
D1 -> RX (pin 2 SDS)
D2 -> TX (pin 1 SDS)

Then, in ext_def.h, I set
```
#if defined(ESP8266)
#define GPS_SERIAL_RX D5
#define GPS_SERIAL_TX D6
#endif
```

closed 2019-04-08 by andreas-bulling