opendata-stuttgart/sensors-software #405: How connect sensor to HomeAssistant?

andreas-bulling, 2019-04-08:
Hi,

I was wondering what the easiest/best way is to get the sensor readings into HomeAssistant. Any standard solutions here (MQTT doesn't seem to be one of them) or people who have been successful with such an integration?

I know that there is a [luftdaten component](https://www.home-assistant.io/components/luftdaten/) but that requires a detour over the internet...

ricki-z, 2019-04-10:
It should be possible to change the mentioned component so that it is using the local json endpoint http://<sensor-ip-address>/data.json . But this endpoint is shut down for some seconds while the firmware is sending data to the external servers (otherwise the NodeMCU will run out of memory).

lichtteil, 2019-04-24:
@andreas-bulling I also have trouble with the luftdaten component (I wrote the very first version of it but it was refactored so much that I have no idea what it is trying to do now :D).
I will now try to use the REST-Component and access the sensor data like this: http://192.168.x.y/data.json

lichtteil, 2019-04-24:
@andreas-bulling I took now my old code of the luftdaten component and changed it so that you can provide the IP address to your local sensor. Feel free to test and give feedback: https://github.com/lichtteil/local_luftdaten

rzr, 2019-06-19:
An other option would be to run a rest to mqtt service on your HomeAssistant host (or elsewhere).

It's trivial with IoT.js:
https://github.com/SamsungInternet/iotjs-express/commit/0ef4de558396f90886bf4d3f28e252298a2ab05e

I can provide more hints if needed.

UFROF, 2019-12-28:
Hi, I also preferred reading the data directly, here my HA conf for that (if you have other sensors attached) look in ../data.json with your browser and change the number in brackets to the correct position number of the value - of course you probably need also to change the IP):
 
sensor:
  - platform: command_line
    name: "Luftdata PM10"
    command: 'curl http://192.168.70.41/data.json'
    value_template: "{{ value_json.sensordatavalues[0].value | round(2) }}"
    unit_of_measurement: "µg/m³"
  - platform: command_line
    name: "Luftdata PM2.5"
    command: 'curl http://192.168.70.41/data.json'
    value_template: "{{ value_json.sensordatavalues[1].value | round(2) }}"
    unit_of_measurement: "µg/m³"
  - platform: command_line
    name: "Luftdata Temperature BME280"
    command: 'curl http://192.168.70.41/data.json'
    value_template: "{{ value_json.sensordatavalues[2].value | round(1) }}"
    unit_of_measurement: "°C"
  - platform: command_line
    name: "Luftdata Humidity BME280"
    command: 'curl http://192.168.70.41/data.json'
    value_template: "{{ value_json.sensordatavalues[4].value | round(1) }}"
    unit_of_measurement: "%"

I assume there is probably even a more elegant way via the rest interface that will probably be able to read the json-date natively but I have not figured it out yet...

beanieboi, 2020-12-29:
I'm using the rest sensor to fetch the values once and then sensor templates to extract all values. This way I can calculate Air Quality Index and a Air Quality Index Description as well.
https://gist.github.com/beanieboi/1c94be3d97bd8bb618f3a31bee12c296

thomasdbnet, 2022-06-06:
thanks beanieboy works perfect. . if not you can always flash the esp to tasmota , it supports the sensor and homeassistant aswel. but it can't be done over the air i think ?