opendata-stuttgart/sensors-software #407: platform.txt

Uwana, 2019-04-24:
Hello,

I'll like to ask, why are we editing the platform.txt file by changing;

'build.float = -u _printf_float -u _scanf_float' 

to

'build.float ='

I just want to understand this a bit more please.

And what is the best way to access the file via arduino?

Thank you.

ricki-z, 2019-04-24:
This compiler switch was important for versions 2.3 and 2.4 of the ESP8266 Arduino core. Otherwise we couldn't get the image size to less than 512 KByte (needed for OTA). We did'nt need the floating point support in printf for the firmware, so this could be disabled to save space.

Even with the actual core 2.5 there is a significant difference im memory usage:
with 'build.float ='
flash image 573880 Bytes (54%)
global vars 35000 Bytes (42%)

with 'build.float = -u _printf_float -u _scanf_float'
flash image 588868 Bytes (56%)
global vars 35020 Bytes (42%)

This is a difference of 15000 Bytes in flash image size.

The location of the file depends on the OS. On Windows it's located in the user directory -> AppData -> Arduino15 -> packages -> esp8266 -> hardware -> esp8266 -> <lib version> . You should edit the file with a normal editor while the IDE isn't running.

ricki-z, 2019-04-28:
Can we close this "issue"?

closed 2019-04-28 by Uwana