opendata-stuttgart/sensors-software #408: LiquidCrystal Compiling Error and Exit

Uwana, 2019-04-26:
Hello,

I get the following error message whenever I run the code in the Arduino IDE:

I have checked to be sure that I have the right library and it is the one recommended for this firmware.

Please guide me on next steps.

Thank you.
::::::::::::::::::::::::::::::

Arduino: 1.8.6 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Disabled, 4M (3M SPIFFS), v2 Lower Memory, Disabled, NoAssert-NDEBUG, Only Sketch, 115200"

WARNING: library LiquidCrystal_I2C claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp8266) architecture(s).
xtensa-lx106-elf-g++: error: {build.led}: No such file or directory

xtensa-lx106-elf-g++: error: {build.led}: No such file or directory

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences._

ricki-z, 2019-04-26:
The Werning can be ignored. It's only a missing entry for the esp8266 in the supported platforms of the LiquidCrystal_I2C lib. We can't change this.
Which esp8266 ardiuno core are you using?

Uwana, 2019-04-26:
Hello Ricki-z

I'm using a Lolin v3.

The first time I compiled, it gave the same warning but went on to compile successfully.

I however couldn't get readings from my SDS011 sensor (I verified that all connections were properly done) but got readings from the DHT22 sensor.

I thought to reload the firmware, but this time it's exiting the compiling process without completion and this is even when I compile on the IDE alone.

Haven't figured out what's gone wrong. I'll appreciate any help.

Thank you.

ricki-z, 2019-04-26:
You could flash a binary flash image. And it seems more a problem with your software installation. The first compile showed only a warning. So the sources should be okay.

Uwana, 2019-04-26:
Alright then, I'll examine from that angle.

Thanks a lot.

closed 2019-04-28 by ricki-z