opendata-stuttgart/sensors-software #409: DS18B20 wird nicht angezeigt.

gerald-b, 2019-05-13:
Hi,

ich verwende die Firmware: **NRZ-2018-123B**
Angeschlossen habe ich "nur"
* OLED SSD1306
* BME280 
* DS18B20

Leider wird mir am OLED Display der DS18B20 nicht angezeigt.
mit dem BME280 passt alles.
Ich habe auch nach dem Anschließen des DS18B20 nochmal das Display "rauskonfiguriert" und anschließend wieder hinzugefügt. Leider ohne Erfolg.
Auch das data.json ist vollständig mit beiden Sensoren

[image]

Was ist die Idee:
Ich möchte den Wasserdichten Sensor (DS18B20) in ein Aquarium hängen.
Der BME280 soll die Temperatur im Zimmer erfassen.
Beide Werte möchte ich am Display ablesen können.

ricki-z, 2019-05-13:
Das liegt jetzt auch als Anwendung etwas neben dem eigentlichen Zweck der Firmware.
Wir zeigen auf dem Display nur den Wert des jeweils "besten" Sensors an. Wenn also z.B. ein DHT22 und ein BME280 verbaut wären, würden die Werte des BME280 angezeigt. Bei einem DHT22 und einem BMP280 (also ohne Luftfeuchte) würde die Temperatur des BMP280, die Luftfeuchte des DHT22 unf der Luftdruck des BMP280 angezeigt.

gerald-b, 2019-06-16:
OK. Danke.

closed 2019-06-16 by gerald-b