opendata-stuttgart/sensors-software #410: airrohr-firmware NRZ 2018-124-B2  - compile error JSON

FrankUlbrich, 2019-05-19:
Hallo,
ich passe für mich die airrohr-firmware gerade für einen WEMOS D1 mini pro an.
Das compilen der Version NRZ 2018-123-B  funktioniert fehlerfrei.
Das compilen der Version NRZ 2018-124-B2 wirrft einen Fehler im Sourcecode 

[image]

Comiled wird mit ARDUINO 1.8.3

Tolles Projekt, habe es zufällig gefunden. Baue an einem Sensor für mich und einen für eine Pfadfindergruppe. 
Danke für die Initiative.
Frank

ricki-z, 2019-05-19:
Hallo,
leider hat die 6er Version der ArduinoJSON Bibliothek einige inkompatible Änderungen. Für die neueste (Beta-)Firmware-Version muss deshalb die Version 6 installiert sein, während für die aktuelle Release Version noch die 5er Version installiert sein muss.

FrankUlbrich, 2019-06-03:
Danke für die Info und vielen Dank für Softwareentwicklung.

closed 2019-06-03 by FrankUlbrich