opendata-stuttgart/sensors-software #411: BME 680 Sensor with Air Quality not supportet?

flinty32, 2019-05-30:
why is BME 680 Air Quality not supportet?

ricki-z, 2019-05-30:
The BME680 doesn't have any additional useable feature compared to the BME280. The VOC measurements are uncalibrated and not compareable between two Sensors in different locations.

imaluschi, 2019-08-21:
The argument concerning calibration is of course true. Nevertheless it makes sense in my opinion. Why? Even if the sensor is uncalibratet you can see changes in gas concentrations. At the moment I'm trying to get an MICS 4541 (NOx and CO Sensor) running on an ESP32. It would be nice to connect it on the ESP8266 (maybe with ADC1115 via I2C) and see the data on the madavi-server.
I want to get data of a MICS 4541 to see wether the NOx and CO concentrations are changing over the week, especially when there is a lot of traffic on the main street - about 50 cars/hour over night, >1200 cars/h on some hours on weekdays, up to >1700 cars/h on saturdays during winter season. How about the air quality? Same, worse?
True, the values can't be compared. But that's not the goal (would be nice, but is not a must).

DeeKey, 2019-12-20:
"measurements are uncalibrated and not comparable between two Sensors in different locations."

What about speed of change? This might be compared even for uncalibrated sensors.
But of course this must be studied in experiment and not for general use..
Yet VOC is not a very certain measurement... Also live of BME680 is not that long...

freddeh, 2020-03-05:
I am looking forward to see a BME680 support. 

FYI, There is already an alternative firmware available from https://forum.fhem.de/index.php/topic,73879.0.html, maybe it makes the implementation faster to have a look?

ricki-z, 2020-03-05:
And again: We won't add support for the BME680.
Think about rooms with constant sources of VOC of different levels (i.e. paints or coatings). Caused by the auto baseline calibration the same minimum value would be shown for all of these rooms.