opendata-stuttgart/sensors-software #412: Feuchtekorrektur / Berechnung der Feinstaubmengen abhänging von der Feuchte

FrankUlbrich, 2019-05-31:
https://github.com/opendata-stuttgart/meta/wiki/Luftfeuchte-Korrektur

Hallo,
einen allgemeine Frage zur Implementierung einer Korrekturrechnung im Sourcecode.
Ist es erlaubt, dass ich meine Werte über eine Formel in Abhängigkeit der Feuchte anpasse und dann hochlade ? Ich kann mir aber auch vorstellen, ds dies nicht erwünscht ist, da nicht zwischen korrigierten und unkorrigierten Werten unterschieden werden kann.
Darf ich oder lieber nicht ? Schöner wäre es allerdings, im json Format Felder für die korrigierten Werte zusätzlich vorzusehen.

MfG
Frank

ricki-z, 2019-06-01:
Wir wollen von den Sensoren ausschließlich die Rohwerte. Sollte ein Fehler in einer Berechnung passieren, können wir so im Nachhinein die Berechnungen neu durchführen.
Die Korrekturrechungen sollten immer auf dem Server (für die dann wieder ausgegebenen, nicht die zu speichernden Daten) erfolgen oder sogar auf dem Client, falls für solche Rechnungen noch "Fremddaten" benötigt werden.
Die Feuchtekorrektur ist nicht ganz so trivial. Hier spielen unter Umständen auch die lokalen Windverhältnisse eine größere Rolle.

FrankUlbrich, 2019-06-03:
Hallo,
danke für die klare Aussage. Ich werde dann weiter Rohdaten senden und meine Rechenexperimente an meiner eigenen Datenbank durchführen.
Gruß
Frank

closed 2019-06-03 by FrankUlbrich